chrippa/ds4drv #52: Game doesn't receive input

CptTeacup, 2015-02-01:
I've tested with Rogue Legacy, Hotline Miami, Nuclear Throne, and Risk of Rain. Rogue Legacy and Hotline Miami receive input without ds4drv running but don't when it is running. Nuclear Throne and Risk of Rain don't receive input whether ds4drv is running or not.

I'm using hidraw with the controller connected with USB, using the default config.

I have tried all three emulation options.

CptTeacup, 2015-02-03:
Tried with xboxdrv instead of ds4drv and it's the same situation.
I guess it has to do with games using the wrong device, but I'm not sure what to do about that.

CptTeacup, 2015-02-04:
Figured it out. Moving the eventX and jsX devices created by ds4drv (or xboxdrv) to the controller's original eventX and jsX devices solves it.

Now I'm wondering why most people aren't having this problem. Should this be handled by ds4drv?

bocajnotnef, 2015-03-05:
I'm also having the issue where Risk of Rain isn't receiving any input--Could you detail the solution you used, please?

bocajnotnef, 2015-03-05:
I'm testing using Steam Big Picture Mode, Bastion and Risk of Rain currently. SBP detects and reacts to joystick events and nothing else.

CptTeacup, 2015-03-06:
I wrote a script that takes care of it, just run this instead of ds4drv.

```
#!/bin/bash

if [[ $(id -u) != 0 ]]; then
    echo "Must be run as root!" >&2
    exit 1
fi

ds4drv --daemon --config /home/$SUDO_USER/.config/ds4drv.conf &
sleep 2

mv /dev/input/event20 /dev/input/event19
mv /dev/input/js1 /dev/input/js0

while pgrep "ds4drv" &> /dev/null; do sleep 1; done

rm /dev/input/event19 /dev/input/js0
```

Make sure the event and js #'s are correct, you want to move the devices created by ds4drv to the original devices that are created when you connect the controller.

Enverex, 2015-05-31:
Same issue here. Steam BPM works fine but games aren't responding. As you mentioned though, why aren't more people seeing this problem?

Enverex, 2015-05-31:
This actually appears to be a duplicate of https://github.com/chrippa/ds4drv/issues/45

Ape, 2016-07-16:
Closed as a duplicate of #45.

closed 2016-07-16 by Ape

Enigma0, 2016-08-29:
@CptTeacup How do you make sure you have the right event/js #'s ?