chrippa/ds4drv #55: installation provlems

Okamooki, 2015-07-30:
Installation fails with this error:

Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-jK6NYC/evdev/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-o3VBQV-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-jK6NYC/evdev

It seems like it might be a compiler error but i don't know how to proceed

Ape, 2015-07-31:
Which Python version? Which OS? What command did you run?

Okamooki, 2015-08-03:
os is ubuntu 14.04,
i do not know how to check what python version pip uses

Ape, 2016-04-30:
Does this still occur?

dotWAVE, 2017-07-05:
I have this same issue trying to sudo pip install ds4drv, running opensuse tumbleweed 20170703. 
pip is version 9.0.1 on python 3.6.

Ape, 2017-07-09:
I'm closing this. For anybody else who might have this issue in the future, try installing python-devel package (or equivalent on your distro).

closed 2017-07-09 by Ape