chrippa/ds4drv #57: Running ds4drv crashes Kodi (XMBC)

loostro, 2015-09-19:
When I run Kodi (15.1 Isengard) first, then run ds4drv it crashes Kodi.
When I run ds4drv first, then Kodi won't start (immidietly crashes).

I've found an issue on Kodi project: [#15826](http://trac.kodi.tv/ticket/15826) it's quite old (7 months) and it seems has not received any attention. Is it possible the problem is on ds4drv side?

If you need any more info, just ask, ill try to deliver any logs/info necessary.

Ape, 2015-09-23:
I think this is most likely a bug in Kodi. Maybe it cannot handle a new joystick device appearing.  It might be that ds4drv uinput device is strange in some sense, but Kodi shouldn't crash in any case.

loostro, 2015-10-01:
Any hints (options) on ds4drv that could possibly make Kodi not crash? Eg. some "safe mode" or "emulate xbox controller"?

ntzrmtthihu777, 2015-12-28:
Similar and perhaps issue occurs with me with steam games and minecraft+liteloader, #45, may be something needed to be fixed here.

Mernst1, 2016-01-17:
you need to use emulate xpad-driver, then kodi will running fine

Ape, 2016-04-30:
Does this still occur?

loostro, 2016-04-30:
I'll test this weekend with new Kodi version.