chrippa/ds4drv #59: multiple controllers

Mernst1, 2015-10-23:
when using multiple controller for example 2.
In Steam Big Picture Mode there is always an input what it makes nearly impossible to navigate in there,
also in games I notice this problem that there is unwanted input. I think always games where you can control the main menu with all four controllers not only with the first.

When I make the config file only with one controller everything works as expected.
My config file:

[ds4drv]
daemon=false
hidraw=false

[controller:1]
led=0000ff
emulate-xpad=true
battery-flash=true
trackpad-mouse=true

[controller:2]
led=ff0000
emulate-xpad=true
battery-flash=true
trackpad-mouse=true

[controller:3]
led=00ff00
emulate-xpad=true
battery-flash=true
trackpad-mouse=true

[controller:4]
led=ffff00
emulate-xpad=true
battery-flash=true
trackpad-mouse=true

[bindings]
PS+Triangle = exec-background killall -KILL pcsxr PCSX2 mupen64plus dolphin-emu snes9x-gtk reicast PPSSPPSDL

vesayth, 2016-02-06:
I'm having this issue as well. Axes 0-6 all have a default minimum value of -32767 which is causing spurious input. This will happen with even a single controller if it has not yet been connected. I've tried to modify the code myself but unfortunately python is not a language that I am comfortable in, and all attempts were for naught.

One thing I have noticed, though, is that these values all get set to 0 when a controller disconnects.

Current workaround is to connect the amount of devices that I'm going to use, but that means manually inputting commands rather than have a script running automatically at boot. This tool would be infinitely better if this problem was solved.

vesayth, 2016-02-08:
Would like to add an update.

Adding the following line to the constructor of UInputDevice in uinput.py fixes the issue:

self.emit_reset()

Of note, it does not change the output of the joystick or jscal tool, but it does get rid of the spurious input that occurs in Steam and other applications.

I suggest anyone that has this problem try this and reinstall ds4drv.

Mernst1, 2016-02-08:
@vesayth 
awesome!!!, thank you that is working for me!!!

Great work. :+1:

benmoran56, 2016-02-09:
It doesn't seem like the original developer is active anymore.  Pull requests are being ignored, and the last commit was 2 years ago.  

There are available lot of useful patches now,  but nobody merging them in.  Maybe it's best if someone can maintain a fork, and accept the patches there.  At least until/if the original developer comes back.

Ape, 2016-04-30:
I merged the fix.

Mernst1, 2016-04-30:
fix doesn't work on new version 0.5.1.
Problem is the same like in 0.5.0.
cant find out why, when I apply the patch on 0.5.0 uinput.py its working.

Ape, 2016-05-04:
@Mernst1 Could you test if #75 with latest master fixes the issue for you?

Mernst1, 2016-05-04:
@Ape  Yes!!! the new fix makes it running again. Great work! thank you @Ryochan7.

closed 2016-05-04 by Ape