chrippa/ds4drv #62: Still relevant?

ntzrmtthihu777, 2015-11-22:
With the hid-sony kernel module adding better support for this device into the kernel itself, is this still relevant for normal usage, eg none of the profiles and xpad and such?

sasann, 2015-11-24:
I only use it to get my controller battery percentage. Any way I can do it without ds4drv?

benmoran56, 2015-11-24:
Yes, battery usage shows up under /sys/class/power_supply/..../capacity. For example, the battery power for my trackpad can be found at: 
/sys/class/power_supply/hid-28\:37\:37\:2e\:07\:82-battery/capacity

sasann, 2015-11-25:
@benmoran56 Awesome tip. Thanks.

samdraz, 2015-12-27:
better useful if this exit as library to other programs

TamasBarta, 2016-01-03:
I have 4.3.3 kernel, and I cannot get my controller to work over bluetooth without ds4drv. I might miss something, but this really makes ds4drv still relevant.

Ape, 2016-04-30:
The plug-and-play kernel driver should work nicely, but ds4drv still have many advanced features compared to it. There are two ways to use ds4drv:

- Raw bluetooth mode (default) provides an alternative way to use the controller. This may be useful in case the kernel driver doesn't work or has some issues.
- Hidraw mode (`--hidraw`) uses the kernel driver and automatically gets all its features. In addition, ds4drv also adds its features like profiles and advances keymappings.

closed 2016-04-30 by Ape