chrippa/ds4drv #63: Duplicate options in config file cause ds4drv to bail out

parkerlreed, 2015-11-24:
````
[parker@lenovo ~]$ sudo ds4drv 
Traceback (most recent call last):
  File "/usr/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "/usr/lib/python3.5/site-packages/ds4drv/__main__.py", line 132, in main
    options = load_options()
  File "/usr/lib/python3.5/site-packages/ds4drv/config.py", line 199, in load_options
    config.load(path)
  File "/usr/lib/python3.5/site-packages/ds4drv/config.py", line 74, in load
    self.read([filename])
  File "/usr/lib/python3.5/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/usr/lib/python3.5/configparser.py", line 1090, in _read
    fpname, lineno)
configparser.DuplicateOptionError: While reading from '/etc/ds4drv.conf' [line 119]: option 'rel_wheeldown' in section 'mapping:keyboard' already exists
````

The ds4drv.conf that comes with git has the options duplicated on line 118

ntzrmtthihu777, 2015-12-28:
I've ran into this same issue. Granted, its very simple to remove the offending lines, but what is in git should work out of box in my opinion.

Ape, 2016-04-30:
I can confirm. Does anybody want to create a fix for this?

closed 2016-04-30 by Ape

Ciruxx, 2019-07-27:
Hello!
Thank's for your job!
I reopen this issue because the exact same error occurred downloading the tarball in the Arch Linux environment!

commit 0860d1d8 (2016-04-30): Merge pull request #74 from mtorromeo/patch-1

Removed duplicate options from default config file

 Fixes #63.
commit cf20ed07 (2016-04-30): Removed duplicate options from default config file (Fixes #63)