chrippa/ds4drv #64: Error when running on Raspbian

x8BitRain, 2016-01-08:
It installs fine but I recieve this error when launching ds4drv http://imgur.com/43N9R5D

I think it has something to do with the packages I have? Though I did install all the dependencies, any ideas?

Mernst1, 2016-01-08:
I can remember I had some similar problems.
The Packages on the Raspbian are a bit differently of the naming if I right.
Try to check that you really have pyudev installed and starting ds4drv with sudo.
I can confirm that ds4drv is working on the raspbian for me,  I noticed when you have overclocked the pi to much you getting lots of connection errors with the driver. The pi is working but the driver is stucking.

ReevMich, 2016-03-03:
There has been a workaround that was posted in another issue similar to this problem. 
https://github.com/chrippa/ds4drv/issues/66#issuecomment-189875102

Ape, 2016-04-30:
Please test if this still occurs with ds4drv 0.5.1.

closed 2020-09-11 by x8BitRain