chrippa/ds4drv #66: can't initialize ds4drv Erno 22

invalidtruck, 2016-02-18:
Can someone tell me how to fix?

`$ ds4drv
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 151, in main
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 120, in create_controller_thread
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 35, in __init__
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 93, in load_options
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 38, in fire_event
  File "build/bdist.linux-x86_64/egg/ds4drv/eventloop.py", line 86, in fire_event
  File "build/bdist.linux-x86_64/egg/ds4drv/eventloop.py", line 92, in process_events
  File "build/bdist.linux-x86_64/egg/ds4drv/actions/input.py", line 79, in load_options
  File "build/bdist.linux-x86_64/egg/ds4drv/uinput.py", line 425, in create_uinput_device
  File "build/bdist.linux-x86_64/egg/ds4drv/uinput.py", line 234, in __init__
  File "build/bdist.linux-x86_64/egg/ds4drv/uinput.py", line 290, in create_device
  File "/usr/local/lib/python2.7/dist-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
IOError: [Errno 22] Invalid argument
`

nitemsg, 2016-02-19:
Having the exact same problem. Setup follows:
debian 8.3 (kernel 3.16.0-4-amd64)
python 2.7.9 (ds4drv 0.5.0 build with this) & 3.4 installed
python-setuptools 5.5.1
bluez 5.23-2+b1 (includes hcitool)
pyudev 0.16.1-2
pip 1.5.6-5
Intel Bluetooth 4.0, 3.0+HS, 2.1+EDR  (motherboard gigabyte z87n-wifi)

colincrossland, 2016-02-22:
Glad to know I'm not the only one getting this problem. Just built up a new system and started running into this, though slightly different due to python 3.4 instead of 2.7

colin@Companion-Cube ~/Downloads $ sudo ds4drv --hidraw --emulate-xpad
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/__main__.py", line 374, in main
    thread = create_controller_thread(index + 1, controller_options)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/__main__.py", line 343, in create_controller_thread
    controller = DS4Controller(index, controller_options, dynamic=dynamic)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/__main__.py", line 59, in __init__
    self.load_options(self.options)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/__main__.py", line 117, in load_options
    self.fire_event("load-options", options)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/__main__.py", line 62, in fire_event
    self.loop.fire_event(event, *args)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/eventloop.py", line 70, in fire_event
    self.process_events()
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/eventloop.py", line 75, in process_events
    callback(*args)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/actions.py", line 282, in load_options
    joystick = create_uinput_device(joystick_layout)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/uinput.py", line 363, in create_uinput_device
    device = UInputDevice(mapping)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/uinput.py", line 227, in __init__
    self.create_device(layout)
  File "/usr/local/lib/python3.4/dist-packages/ds4drv/uinput.py", line 268, in create_device
    product=layout.product, version=layout.version)
  File "/usr/local/lib/python3.4/dist-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
SystemError: ../Objects/longobject.c:404: bad argument to internal function

so now the revision history hunt begins!

t0mfra, 2016-02-23:
I have the same problem (both python2.7 and 3.4)

sudo ds4drv
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/__main__.py", line 151, in main
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/__main__.py", line 120, in create_controller_thread
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/__main__.py", line 35, in __init__
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/__main__.py", line 93, in load_options
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/__main__.py", line 38, in fire_event
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/eventloop.py", line 86, in fire_event
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/eventloop.py", line 92, in process_events
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/actions/input.py", line 79, in load_options
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/uinput.py", line 425, in create_uinput_device
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/uinput.py", line 234, in __init__
  File "/usr/local/lib/python3.4/dist-packages/ds4drv-0.5.0-py3.4.egg/ds4drv/uinput.py", line 290, in create_device
  File "/usr/local/lib/python3.4/dist-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
SystemError: ../Objects/longobject.c:404: bad argument to internal function

pasmai, 2016-02-24:
I also have this issue on ubuntu.
Ubuntu 15.10
python 2.7.10
pip 1.5.6
bluetoothctl 5.35

`Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 151, in main
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 120, in create_controller_thread
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 35, in __init__
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 93, in load_options
  File "build/bdist.linux-x86_64/egg/ds4drv/__main__.py", line 38, in fire_event
  File "build/bdist.linux-x86_64/egg/ds4drv/eventloop.py", line 86, in fire_event
  File "build/bdist.linux-x86_64/egg/ds4drv/eventloop.py", line 92, in process_events
  File "build/bdist.linux-x86_64/egg/ds4drv/actions/input.py", line 79, in load_options
  File "build/bdist.linux-x86_64/egg/ds4drv/uinput.py", line 425, in create_uinput_device
  File "build/bdist.linux-x86_64/egg/ds4drv/uinput.py", line 234, in __init__
  File "build/bdist.linux-x86_64/egg/ds4drv/uinput.py", line 290, in create_device
  File "/usr/local/lib/python2.7/dist-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
IOError: [Errno 22] Invalid argument`

Hadatko, 2016-02-28:
I am no aloooone. I have this issue on raspberry pi jessie. When i had raspberry pi wheezy and i upgraded to jessie it worked. But after clean install of jessie it is not :/ Still nobody know, how to resolve this?

MrPepperoni, 2016-02-28:
Not a fix, but a workaround, that solved the issue for me:
downgrade evdev to 0.5.0
`sudo pip install evdev==0.5.0`
if you have a kernel version >= 4.4.0, apply the patch described here: https://github.com/gvalkov/python-evdev/issues/43#issuecomment-177664399
`sudo pip install --upgrade git+https://github.com/gvalkov/python-evdev.git@631e2d32d7bdf38e3d7a5c850c9f5869d61e9183`

Hadatko, 2016-02-28:
You are king ;) Your workaround works perfectly.

hjsav, 2016-03-05:
I applied this workaround and it connects and automatically disconnects. 
[image]

Ape, 2016-04-30:
I think this might be fixed in ds4drv 0.5.1. Please test with it and report if this still occurs.

metanerd, 2016-05-09:
This issue didn't occur again for me after I piped the developer version on the stable one. 

My ds4 though does not get connected. It is also not powered on. Pretty sure I am missing something and/or using this tool wrongly.

> [info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event17 (evdev) 
> [info][bluetooth] Scanning for devices

multiplegeorges, 2016-05-12:
Rasbian Jessie here, kernel 4.4.

ds4drv 0.5.1 works perfectly here.

superjamie, 2016-05-22:
Ubuntu 16.04, kernel 4.4, also fixed in ds4drv 0.5.1

closed 2016-05-22 by Ape

IrigoyenCarlosDamian, 2022-09-01:
when I run the following command I get the following error
python3 -m ds4drv --hidraw --udp
[info][controller 1] Created devices /dev/input/js2 (joystick) /dev/input/event259 (evdev) 
[info][controller 1] Connected to USB Controller (A4:53:85:21:1A:22 A4:53:85:21:1A:22 hidraw8)
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/__main__.py", line 222, in <module>
    main()
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/__main__.py", line 219, in main
    thread.controller.setup_device(device)
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/__main__.py", line 79, in setup_device
    self.device.set_led(*self.options.led)
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/device.py", line 98, in set_led
    self._control()
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/device.py", line 87, in _control
    self.control(led_red=self._led[0], led_green=self._led[1],
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/device.py", line 148, in control
    self.write_report(report_id, pkt)
  File "/home/carlos/.local/lib/python3.10/site-packages/ds4drv/backends/hidraw.py", line 67, in write_report
    self.fd.write(hid + data)
TimeoutError: [Errno 110] Connection timed out

my specification
 linux mint 21 
python 3.10 
 Kernel: 5.15.0-46