chrippa/ds4drv #69: Failed to create input

hjsav, 2016-03-05:
every time I install i can't get it to install. This is what i get when I run the command.   

$ ds4drv
[error][controller 1] Failed to create input device: "/dev/uinput" cannot be opened for writing
[info][bluetooth] Scanning for devices

please help and thank you.

MrPepperoni, 2016-03-05:
Probably ds4drv doesn't have permissions to access /dev/uinput, please check https://github.com/chrippa/ds4drv#permissions
You can also try using --hidraw if you have bluez 5.14+, I found https://wiki.gentoo.org/wiki/Sony_DualShock#DualShock_4 to be the most straightforward way to pair the controller.

Ape, 2016-04-30:
Is this still an issue?

CoryParsnipson, 2016-06-15:
I had the same problem, and I found [this page](https://groups.google.com/forum/#!topic/ds4drv/CdPZm4EZRMM) that has the answer, I think.

I did not have uinput loaded by default. That is at the time, lsmod | grep uinput returns nothing. If I load uinput ds4drv seems to be able to create /dev/input/js* just fine.

To make uinput start at boot, you can add a file, /etc/modules-load.d/uinput.conf, for example with these contents:

`/etc/modules-load.d/uinput.conf
--------------------------------------------------
# Load uinput module at boot
uinput`

As Joseph says in the thread I linked.

Ape, 2016-06-15:
Thanks for the solution. This is not really a bug in ds4drv so I am closing this. Please submit a pull request if you have ideas about how to improve the documentation or error printing.

CoryParsnipson, 2016-06-17:
Ok, thanks. I'll probably see if I can add a little section about uinput underneath the udev rules thing.

commit 3f1fa743 (2016-06-20): Add troubleshooting section to readme for uinput related error

Resolves #69.
closed 2016-06-20 by Ape