chrippa/ds4drv #70: Fail on start with python-evdev 0.6.0

Ape, 2016-03-21:
Previously ```ds4drv``` worked correctly with ```python-evdev``` 0.5.0. I upgraded ```python-evdev``` to version 0.6.0 and now ```ds4drv``` fails on start with this:

```
Traceback (most recent call last):
  File "/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "/usr/lib/python3.5/site-packages/ds4drv/__main__.py", line 374, in main
    thread = create_controller_thread(index + 1, controller_options)
  File "/usr/lib/python3.5/site-packages/ds4drv/__main__.py", line 343, in create_controller_thread
    controller = DS4Controller(index, controller_options, dynamic=dynamic)
  File "/usr/lib/python3.5/site-packages/ds4drv/__main__.py", line 59, in __init__
    self.load_options(self.options)
  File "/usr/lib/python3.5/site-packages/ds4drv/__main__.py", line 117, in load_options
    self.fire_event("load-options", options)
  File "/usr/lib/python3.5/site-packages/ds4drv/__main__.py", line 62, in fire_event
    self.loop.fire_event(event, *args)
  File "/usr/lib/python3.5/site-packages/ds4drv/eventloop.py", line 70, in fire_event
    self.process_events()
  File "/usr/lib/python3.5/site-packages/ds4drv/eventloop.py", line 75, in process_events
    callback(*args)
  File "/usr/lib/python3.5/site-packages/ds4drv/actions.py", line 282, in load_options
    joystick = create_uinput_device(joystick_layout)
  File "/usr/lib/python3.5/site-packages/ds4drv/uinput.py", line 363, in create_uinput_device
    device = UInputDevice(mapping)
  File "/usr/lib/python3.5/site-packages/ds4drv/uinput.py", line 227, in __init__
    self.create_device(layout)
  File "/usr/lib/python3.5/site-packages/ds4drv/uinput.py", line 268, in create_device
    product=layout.product, version=layout.version)
  File "/usr/lib/python3.5/site-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
SystemError: Objects/longobject.c:403: bad argument to internal function
```

theredbaron1834, 2016-03-28:
Hitting that issue as well on Arch Linux. Can't find a way around it.

Muskipet, 2016-03-31:
Same problem on Manjaro/Archlinux. Maybe the workaround is to get back on the 0.5.0 version of python-evdev, but i don't know how to do on Archlinux.

Ape, 2016-03-31:
@Muskipet https://wiki.archlinux.org/index.php/downgrading_packages

theredbaron1834, 2016-04-15:
@Ape The problem with that is that ds4drv on install notices that there is a new evdev, and installs it as a depend. I can't see a way to block this from happening, or to force it to use an older installed version. I am sure it is easy enough, but I know nothing about Python, so out of my league.

Ryochan7, 2016-04-17:
It looks like the following change to python-evdev broke compatibility with ds4drv. There is now an extra index for absinfo corresponding to the current value for an axis. The new representation used for uinput device creation has a size of 6 while the one made from the data passed by ds4drv has a size of 5. Adding new items at index 1 for each absinfo entry (using 0 for testing) allows ds4drv to work.

https://github.com/gvalkov/python-evdev/commit/62969d7bca43d87799237590a460d4361012225c#diff-4837c124766bc823c51073050068c3ee

Also relevant: https://github.com/gvalkov/python-evdev/blob/master/evdev/device.py#L20

Ape, 2016-04-17:
@Ryochan7 Could you send a fixing pull request to ds4drv?

closed 2016-04-30 by Ape

Ape, 2016-04-30:
I released version 0.5.1 which works with python-evdev 0.6.0.

commit 2bb335ad (2016-04-30): Add value item to absinfo tuples

Needed to make ds4drv compatible with python-evdev
0.6.0.

Resolves #70.