chrippa/ds4drv #71: Signal strength issues on Raspberry Pi3 with DS4

lexwbr, 2016-04-13:
Hi,

I am using a Raspberry Pi3 with internal Bluetooth and RetroPie 3.6 installed. After installation as described in http://blog.petrockblock.com/forums/topic/tutorial-retropie-with-dualshock-4-and-bluetooth/
I can pair the Dualshock 4 controller, but I am having major lags compared to connecting it via USB. Here is the output which also confirms the lag by `[warning][controller 1] Signal strength is low (21 reports/s)`

```
pi@retropie:~ $ ds4drv --led 000008
[info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event0 (evdev)
[info][bluetooth] Scanning for devices
[info][bluetooth] Found device D0:27:88:5F:36:15
[info][controller 1] Connected to Bluetooth Controller (D0:27:88:5F:36:15)
[info][bluetooth] Scanning for devices
[info][controller 1] Battery: 37%
[warning][controller 1] Signal strength is low (15 reports/s)
[warning][controller 1] Signal strength is low (21 reports/s)
[warning][controller 1] Signal strength is low (19 reports/s)
[info][controller 1] Battery: 25%
[warning][controller 1] Signal strength is low (26 reports/s)
```

Is there something I could do in order to increase the signal strength?

closed 2016-04-13 by lexwbr

Ape, 2016-04-14:
Did you find a solution?

lexwbr, 2016-04-14:
@Ape no, I just stopped using ds4drv. RetroPie 3.6 is able to recognize a DS4 on its own and without any lag issues, in contrast to ds4drv.