chrippa/ds4drv #76: Information about playing audio on DS4

a1ien, 2016-06-01:
I have some research to be able playing audio on DS4.
Audio data send by opcode 0x14, 0x15, 0x17, 0x19.
And code by SBC audio codec.
I used this python script to wrap the audio data
play.py
This script takes the SBC audio data and send it to DS4
You can use gstreamer to prepare audio data.
gst-launch-1.0 -q audiotestsrc ! sbcenc ! 'audio/x-sbc,channels=2,rate=32000,channel-mode=dual,blocks=16,subbands=8,bitpool=25' ! queue ! fdsink | ./play.py
But there is a small problem, periodically occur stuttering. But it's maybe
But it may be caused by improper operation with HID device. 
I hope this helps to add audio support.

Ape, 2016-06-02:
Wow! This is actually working. Thank you so much for this.

I also get some stuttering. Perhaps the bluetooth signal is not perfect.

a1ien, 2016-06-05:
And small addition
If add is-live=true to audiotestsrc
`gst-launch-1.0 -q audiotestsrc is-live=true ! sbcenc ....`
Stuttering becomes much less
And useful pipeline to play all output to DS4
`gst-launch-1.0 -q pulsesrc device="alsa_output.pci-0000_00_1b.0.analog-stereo.monitor" ! queue ! audioresample ! 'audio/x-raw,rate=32000' ! audioconvert ! sbcenc ! 'audio/x-sbc,channels=2,rate=32000,channel-mode=dual,blocks=16,subbands=8,bitpool=25' ! queue ! fdsink | ./play.py`
To get device name u can use
`pacmd list-sources | grep -e device.string -e 'name:'`

Ape, 2016-06-05:
Great! Do you know if there is a way to control the volume?

a1ien, 2016-06-05:
Yes in last script version have variables
volume_l, volume_speaker.

parkerlreed, 2016-06-06:
I have mine plugged in via USB without ds4drv running

dmesg shows the controller as /dev/hidraw5 and the script is set to use it

````
        name: <alsa_output.pci-0000_00_1b.0.analog-stereo.monitor>
````

````
gst-launch-1.0 -q pulsesrc device="alsa_output.pci-0000_00_1b.0.analog-stereo.monitor" ! queue ! audioresample ! 'audio/x-raw,rate=32000' ! audioconvert ! sbcenc ! 'audio/x-sbc,channels=2,rate=32000,channel-mode=dual,blocks=16,subbands=8,bitpool=25' ! queue ! fdsink | ./play.py
````

Without the play.py, fddsink outputs data to the terminal correctly. With play.py the controller vibrates sporadically and has no sound coming from the speaker.

Arch x86_64

a1ien, 2016-06-06:
Hmm... It's strange.
You can test 
`gst-launch-1.0 -q audiotestsrc is-live=true ! sbcenc ! 'audio/x-sbc,channels=2,rate=32000,channel-mode=dual,blocks=16,subbands=8,bitpool=25' ! queue ! fdsink | ./play.py`
And what Kernel version you have?
`uname -a`
I test on last ubuntu 16.04

parkerlreed, 2016-06-06:
audiotestsrc was also silent

````
Linux e73 4.5.4-1-ARCH #1 SMP PREEMPT Wed May 11 22:21:28 CEST 2016 x86_64 GNU/Linux
````

a1ien, 2016-06-06:
>EDIT: Is this supposed to only work via Bluetooth?
Yes only Bluetooth

poconbhui, 2016-06-08:
This is working for me. I used `pactl load-module module-null-sink sink_name=ds4` to set up a nicer looking pulseaudio sink, which audio outputs can be selectively output to from `pavucontrol`. The `pulseaudio-dlna

parkerlreed, 2016-06-08:
What did you pass to gstreamer to make it use the new sink?

Ape, 2016-06-08:
@parkerlreed You can use this to get the names:
```
pacmd list-sources | grep -e device.string -e 'name:'
```

So if your named sink is `ds4` then you can probably use this:

```
gst-launch-1.0 -q pulsesrc device="ds4.monitor"
```

parkerlreed, 2016-06-08:
Ahh thanks.

poconbhui, 2016-06-08:
If you open `pavucontrol`, under the "playback" tab, you can select the "Null Output" output from a dropdown menu once you have something playing.

poconbhui, 2016-06-08:
The 16000 does sound a lot crappier though.

a1ien, 2016-06-08:
You also can try always send _17_report instead _14_report. For the 32kHz, works better.

poconbhui, 2016-06-08:
So report 17 and 14 are the same except for the amount of audio data? I'm getting much better playback having replaced everything with report 17.

The hidraw writing freezing every once in a while is still a problem for me though. Is anyone else having that problem?

I'm sort of getting around it by setting a SIGALRM timeout as described at http://stackoverflow.com/a/22348885/445799 . Not the nicest option though.

a1ien, 2016-06-08:
>So report 17 and 14 are the same except for the amount of audio data?

Yes. Also [19](http://eleccelerator.com/wiki/index.php?title=DualShock_4#0x19) same as 15 except for the amount of audio data. 

And first part of 15 and 19 same as 11

poconbhui, 2016-06-08:
Nice. Any info on the timing between reports? I'm finding I have better sound with a slight delay between packets. For me, it's better when I print a bunch of garbage to the terminal than when I don't.

a1ien, 2016-06-08:
>Any info on the timing between reports?

No, but http://eleccelerator.com/files/ds4_uart_hci_cap_playroom_needs_sorting.pcap.gz may be help.

You cant filter packet from PS4 using bthid[0] == 0xa2
[img](https://habrastorage.org/files/acf/07f/db8/acf07fdb8ee64554b01c1eddc5587e4d.png)

poconbhui, 2016-06-08:
Any recommendations on programs to use to look at that file? I'm not too familiar with hardware stuff, so I've not touched something like this before.

Ok, I'm having better luck setting up a SIGALRM timeout in a `multiprocessing.Process`. I was using a `threading.Thread`, but performance wasn't good enough. Setting the timeout to somewhere around 2*448/32000 seconds is working well enough to allow for a dropped packet or two. There must be a better solution though.

Moving controllers in ds4drv out to processes might be a good idea anyway.

I've added the changes to your play.py with the timeout process [here](https://gist.github.com/poconbhui/ab3ea6c5b827c969bd0cc8d76575a0a2)

a1ien, 2016-06-08:
Wireshark can open file. You can see on screenshot. Before open you need ungzip.

poconbhui, 2016-06-08:
Woops, forgot about the encoding. Testing with dd suggests data is coming in at about 27.9 kB/s when encoding a 32 kB/s stream.

poconbhui, 2016-06-08:
I think I've found the volume balance. In `b'\x02'+ audo_data` of the reports, it's that `\x02` value.

a1ien, 2016-06-08:
> I think I've found the volume balance

What means "volume balance". For what purpose use \x02 before audio_data?

poconbhui, 2016-06-08:
When playing stereo audio, only one channel was coming through and my right earphone was very quiet.

Setting it to different values changes the relative loudness of the two earphones. So it seems there might not be a left volume / right volume, but instead a total volume and relative volume.

a1ien, 2016-06-08:
>When playing stereo audio, only one channel was coming through and my right earphone was very quiet.

I ran into the same problem, but take earphone with microphone on the jack I get right sound good.

It seems that only supports mono support.

poconbhui, 2016-06-08:
This seems to be working for me:
`header = bytes([int('00100011', 2)])`, with header in place of \x02.

poconbhui, 2016-06-08:
I was looking at https://tools.ietf.org/html/draft-ietf-avt-rtp-sbc-01#section-6.3

poconbhui, 2016-06-08:
Looks like `00100100` should be the right header for report 17 for the sbcenc settings from gstreamer.

a1ien, 2016-06-08:
No. Gstreamer generate sbc header.
0x02 - it's not SBC header
SBC header is 0x9c, 0x75, 0x19, 0x24
0x9c  - sync 
0x75 - 0b1110101
0x19 - bitpool
...

poconbhui, 2016-06-08:
Sure, it generates the SBC header, but it doesn't generate the payload header. Each SBC frame with the current settings is 112 bytes long, so each report sent to the ds4 contains 4 SBC frames, each with an SBC frame header generated by gstreamer, but those 4 have their own header, the payload header, where `\x02` is, which should be `00100100`, following section 6.1 of the sbc doc I linked. Gstreamer can't generate this header because it doesn't know how many SBC frames are going to be put in a payload.

The only two things that seem to make a difference is setting the L bit to 0 causes the right channel to be very quiet and not setting the \#frames to something > 1 causes the sound to jitter, or just not work.

a1ien, 2016-06-08:
Oh sure, I did not notice section 6.1

poconbhui, 2016-06-08:
With some of that info, it might be possible to get the mic working as well. I found the length of the SBC frames by looking for the syncword  `\x9c` and looking for the most common spacing. Looking for regularly spaced `\x9C`s in the input reports could show it up. The stream might then be decoded following the SBC frame header spec.

a1ien, 2016-06-08:
For 15 report \x02 seems that there should be also 00100100(\x24), but why?
15 report contain only 2 frame.

a1ien, 2016-06-08:
volume_unk1 - it's volume_r when header = 0x24
If header = 0x24  speaker not work
If header = 0x02  speaker work

poconbhui, 2016-06-09:
It might not be the payload header then... I noticed with 0x02 my left earphone worked, my right was pretty quiet and there was one loud and one quiet channel coming out of the speaker.

poconbhui, 2016-06-09:
Playing about with those bits, the main difference seems to be setting the 3rd bit. 1 means the right channel goes to the earphone and volume is controlled by volume_r, 0 means the right channel goes to the speaker and is controlled by volume_speaker. The left ear is probably muted and the audio made mono when outputting to the speaker.

poconbhui, 2016-06-09:
I think I found what was going wrong with the timeout I was using. I was looking for problems on the wrong thread. My version of the script seems to be playing audio perfectly now.

I've been looking over the SBC specs and I'm working on parsing the headers and automatically loading the right number of bytes from stdin. Specs are at https://developer.bluetooth.org/TechnologyOverview/Pages/A2DP.aspx , the pdf is A2DP_v1.3.1.pdf, and the frame length stuff is in section 12.9.

a1ien, 2016-06-09:
What distro you use? I ask because on Ubuntu 16.04 I don't need configure any timeout.
My version work without shuttering after set header to 0x24 even send 14 report.
Also 'audio/x-sbc,channels=2,rate=32000,channel-mode=**stereo**,blocks=16,subbands=8,bitpool=**50**'
sound good.
Linux 4.4.0-22-generic #39-Ubuntu SMP Thu May 5 16:53:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
And this my kernel.

poconbhui, 2016-06-09:
I'm on Arch Linux
Linux skuzylbutt 4.5.4-1-ARCH #1 SMP PREEMPT Wed May 11 22:21:28 CEST 2016 x86_64 GNU/Linux

It usually sounds fine, but every once in a while writing to the hidraw interface freezes.

poconbhui, 2016-06-09:
Playing around with the SBC settings, there only seem to be a couple of allowed combinations of values. Have you gotten it playing audio at 44100 or 48000? According to the SBC specs, it's supposed to support at least one.

a1ien, 2016-06-09:
Yes I try  44100 and 48000, not work (
At least case, I couldn't make it work

poconbhui, 2016-06-09:
I just sent a mono encoding to it and it came out the speaker instead of the ear phones.

poconbhui, 2016-06-09:
On mono, it seems to be bit 4 of the audio header that determines whether it comes out the earphone or the speaker, but I'm only getting the left earphone working.

a1ien, 2016-06-09:
0x22 seem for channel-mode=joint

ntzrmtthihu777, 2016-06-10:
So, here's a question. Since there is now an in-tree kernel module, would it be possible just present the ds4 as a true audio device in that manner?

poconbhui, 2016-06-10:
We're probably doing a lot of unnecessary stuff that could be handled by the kernel tree module. I've been having too much fun playing with this to give it a proper look.

a1ien, 2016-06-10:
We must somehow encode to SBC in kernel. And this is a problem.

poconbhui, 2016-06-10:
I've been looking into SBC encoding to try and remove the gstreamer pipe and do all the audio stuff in python as well, since I'm looking to add this nicely to ds4drv once I've got my kinks worked out.

Just had a look at the kernel driver code https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/drivers/hid/hid-sony.c?id=refs/tags/v4.5.7 and there doesn't seem to be any audio stuff in here.

poconbhui, 2016-06-10:
Yeah, there's also an SBC encoder that ships with bluez, which is probably installed on a linux system using bluetooth. I'm looking into pulseaudio programming as well. I'm not sure how widespread gstreamer is and the SBC encoder is in a separate gstreamer plugins package on my system, which is why I'm looking at how viable replacing it is.

a1ien, 2016-06-10:
SBC plugin come with gstreamer1.0-plugins-bad.
And without gstreamer(pure pulseaudio) you have many problems with frame rate and format. Also, not everyone uses pulseaudio.
I do not see any problems with the use of gstreamer, he is in all linux distributions.
And even in windows.

poconbhui, 2016-06-10:
There's no problem, really. It works just fine, and it's readily available. I'm sort of interested in the audio stuff for a separate project. and if it's easy enough to get rid of one or two extra dependencies, then good.

Ape, 2016-06-10:
We can make these dependencies optional and only enable audio support on ds4drv is they are present.

parkerlreed, 2016-06-10:
Is audiotestsrc still working for anybody? I just did a fresh Arch install because by Bluetooth crapped out. I got the controller connected again but when I run the gstreamer to play.py pipe the only thing that happens is the blue light on the controller goes from a dull blue to a brighter blue.

poconbhui, 2016-06-11:
audiotestsrc only works for me with `is-live=true` set.

poconbhui, 2016-06-11:
@Ape of course. I'm just being funny. I'm working on a gstreamer pipeline and thinking about the best way to incorporate it in the current event loop. I'll do a little experimenting, but it might be an idea to have two threads per controller, one for input, one for output.

poconbhui, 2016-06-11:
I got gstreamer running in python following http://brettviren.github.io/pygst-tutorial-org/pygst-tutorial.pdf, but the basic model is a bit gross. It involves setting up a blocking GTK main loop. This can probably be pulled off to another thread. I'm not a big fan of it, so I'm going to see if there are any other options.

a1ien, 2016-06-11:
@parkerlreed last  play.py output only to headset.
You can replace header = b'\x24' to header = b'\x02' for play from speaker

poconbhui, 2016-06-11:
Ok, I have a working gstreamer version at https://gist.github.com/poconbhui/ab3ea6c5b827c969bd0cc8d76575a0a2 .

poconbhui, 2016-06-11:
Should probably put the gstreamer buffer filters back in.

Probably don't need the SBC frame header parser in there anymore because we should already know what the SBC parameters are. Gstreamer might change the values we've asked for though, I have no idea.

The stuff I was using for file write timeouts aren't working with this because of the write not happening in the main thread or something like that. I'm looking into moving it to another process.

[17 more replies]