chrippa/ds4drv #83: Invalid Python 2 syntax in __main__.py

Ape, 2016-07-02:
When running `ds4drv` with Python 2 on the master branch I get this error:

```
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.1', 'console_scripts', 'ds4drv')()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 351, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2363, in load_entry_point
    return ep.load()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2088, in load
    entry = __import__(self.module_name, globals(),globals(), ['__name__'])
  File "/usr/local/lib/python2.7/dist-packages/ds4drv-0.5.1-py2.7.egg/ds4drv/__main__.py", line 112
    def exit(self, *args, error = True):
                              ^
SyntaxError: invalid syntax
```

I usually use Python 3 and there is no error there, but Python 2 should be supported, too.

This is most likely caused by [this commit](https://github.com/chrippa/ds4drv/commit/63ff9927fedd3742a5719e0315a63eabce59fa5a). @poconbhui please see this.

poconbhui, 2016-07-02:
Oh butts. Should be quick enough to fix when I get a chance.

poconbhui, 2016-07-03:
Should be fixed in #86.

PR #86 (2016-07-03): Fixed DS4Controller.exit signature for python 2.
Fixes #83 for python2.7.
closed 2016-07-03 by Ape