chrippa/ds4drv #84: Updating default config

Ape, 2016-07-02:
I was thinking if we should modify the default configuration in some ways. I'd like to hear your opinions and comments on this subject. I will list the changes that could be made. Please tell if there are more changes you would like to propose.

1. **Hidraw mode as default**
  - The raw / direct bluetooth mode is the default because back in the early days there was no kernel support for the controllers, so it was the only possible mode. However, modern distributions have newer versions of Linux kernel and Bluez, so hidraw works.
  - I think hidraw mode is fundamentally better and with the latest commits and pending pull requests it will support all features in ds4drv. It also supports USB mode.
  - This is breaking change for people that do not specify the hidraw mode in their current configuration. And it might be that hidraw doesn't work for everybody, and some people may prefer the temporary connection to actually pairing the controllers.
  - I will definitely want to everybody's opinion on this one. Do you prefer to use hidraw or raw bluetooth?

2. **Trackpad mouse enabled on controller 1**
  - I think most people would prefer if trackpad mouse was enabled by default on the first controller. It can be disabled in the configuration.

3. **Battery flash by default**
  - Battery flash could be enabled on all controllers by default.

4. **Emulate xpad by default**
  - Most games are made for the Xbox 360 controller and seems that `emulate-xpad` works with most games as intended.
  - However, some games might actually support DualShock 4, and this could break that support.
  - The original DS4 mode could be a default profile in the default configuration.

5. **Mouse sensitivity 0.8**
  - This is just a matter of preference, but I think mouse sensitivity `0.8` is better than `0.6`.

6. **Mouse deadzone 4**
  - The current default mouse deadzone is `5`, and it may be the best value. For my controller I think `4` also works without problems. However, some controllers might have issues with the value `4`. I'd like to hear if `4` works for you, or if the default `5` should be kept.

poconbhui, 2016-07-03:
1. I think hidraw provides more features, but explicit pairing might be a little awkward for users. It took me a little while to figure out, and I still didn't get any GUI bluetooth tools working. Maybe I did a dumb though. Is there any way ds4drv could manage pairing and creating a hidraw interface?

4. I usually go with xpad or keyboard mappings. Actually, one thing I found frustrating is that I can't specify emulate-xpad and then change *some* buttons, like xpad for most things and set the left analog to the mouse. I looked into it and fixing it would be a little gross, but doable. Do you think it would be worth doing?

  Also what is the "default" profile? When setting profiles from profiles=x,y,z, the default should be disabled unless explicitly passed.

5. I found I had to increase my mouse sensitivity playing Portal 2, so I'd agree with higher mouse sensitivity.

6. I find my mouse still sometimes trails off and I need to flick the analog stick to stop it. I think it's fairly controller dependant. I'll try setting it to 4 and see if it's any worse.

I was thinking something like a splash screen might be nice. Like flash the LEDs a little and rumble the pad to show it's been connected. Then add an option in the configs to turn that off.

Ape, 2016-07-03:
1. The pairing can be done in multiple ways. For example there is a plugin in Bluez that automatically pairs the controller when it is plugged in with USB. But it might be possible to implement a pairing assistant that calls bluetoothctl and guides the user. I agree that pairing should be easier before hidraw can become the default. Let's discuss this in #88.

Ape, 2016-07-03:
@poconbhui 2. Yes, the mapping configuration could use some rethinking.

By the default profile, I meant the mappings that are used by default if the user does not explicitly configure it. So should the controller behave like a DS4 does with plain kernel drivers (no ds4drv) or like a Xbox 360 controller (assumed by many games) if the user doesn't change the configuration at all?

Ape, 2016-07-03:
@poconbhui 4. Are you having those problems with the deadzone setting `5`? So maybe it should be increased in the default configuration instead. Try what is the lowest value for you that is never causing any problems. For me even `4` never causes any trailing.

Enigma0, 2016-08-28:
Should definitely have `hidraw` enabled by default and `xpad/xboxdrv` enabled by default. Having trouble myself getting the controller to work with anything other than Steam Big Picture (no games) - https://github.com/chrippa/ds4drv/issues/97.

Also definitely have the touchpad enabled by default. Pretty awesome feature.