chrippa/ds4drv #85: Two finger scrolling

Ape, 2016-07-02:
I think we can read two touch points at the same time, so two finger scrolling could be possible to implement. We could just send scroll button events, but I would prefer a smooth fine grained scrolling that is supported in many laptop touch pads.

poconbhui, 2016-07-03:
I already use two finger scrolling with ds4drv.

    REL_WHEEL  = trackpad_touch1_x                                                   
    REL_HWHEEL = trackpad_touch1_y

Ape, 2016-07-03:
@poconbhui Thanks! I think `REL_WHEEL = trackpad_touch1_y` works nicely. Do you know if it's possible to invert the direction?

But we should still implement proper two finger scrolling. `touch1` means the second finger. So the first finger still controls the mouse and the second finger controls scrolling at the same time. That works quite nicely, but it's not as good as real two finger scrolling on most laptops.

We should detect if both fingers are moving together and then use the average for scrolling. The mouse cursor should not move when doing a two finger scroll. Also, we should research if it possible to do the more fine grained scrolling instead of mouse wheel button events.

poconbhui, 2016-07-04:
Does `REL_WHEEL = -trackpad_touch1_y` work?

I wonder if it's possible to tap into the synaptics driver for this? Looking at it's default options and how it implements them would be a good idea at least.

Things might be a little more complicated than just using `REL_WHEEL` for things like pinch zoom.

Also, just found https://www.kernel.org/doc/Documentation/input/event-codes.txt . There's certainly some stuff in here we've not been doing regarding the mouse pad.