chrippa/ds4drv #95: error when: ds4drv --hidraw

Habacuque, 2016-07-21:
Hello
i am using debian jessie in a computer without bluetooth
but when (controller via usb) i write  ds4drv --hidraw(as root) i get this:

"Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 9, in <module>
    load_entry_point('ds4drv==0.5.0', 'console_scripts', 'ds4drv')()
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 374, in main
    thread = create_controller_thread(index + 1, controller_options)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 343, in create_controller_thread
    controller = DS4Controller(index, controller_options, dynamic=dynamic)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 59, in __init__
    self.load_options(self.options)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 117, in load_options
    self.fire_event("load-options", options)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 62, in fire_event
    self.loop.fire_event(event, *args)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/eventloop.py", line 70, in fire_event
    self.process_events()
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/eventloop.py", line 75, in process_events
    callback(*args)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/actions.py", line 282, in load_options
    joystick = create_uinput_device(joystick_layout)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/uinput.py", line 363, in create_uinput_device
    device = UInputDevice(mapping)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/uinput.py", line 227, in __init__
    self.create_device(layout)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/uinput.py", line 268, in create_device
    product=layout.product, version=layout.version)
  File "/usr/local/lib/python2.7/dist-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
IOError: [Errno 22] Invalid argument"

Ape, 2016-07-22:
This should be fixed in ds4drv 0.5.1. Can you confirm this? See #66.

walchko, 2016-07-28:
Hello, I just installed this today, but it was 0.5.0 and got this error (same as above):

```bash
pi@zoidberg:~/tmp $ ds4drv 
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 374, in main
    thread = create_controller_thread(index + 1, controller_options)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 343, in create_controller_thread
    controller = DS4Controller(index, controller_options, dynamic=dynamic)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 59, in __init__
    self.load_options(self.options)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 117, in load_options
    self.fire_event("load-options", options)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/__main__.py", line 62, in fire_event
    self.loop.fire_event(event, *args)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/eventloop.py", line 70, in fire_event
    self.process_events()
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/eventloop.py", line 75, in process_events
    callback(*args)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/actions.py", line 282, in load_options
    joystick = create_uinput_device(joystick_layout)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/uinput.py", line 363, in create_uinput_device
    device = UInputDevice(mapping)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/uinput.py", line 227, in __init__
    self.create_device(layout)
  File "/usr/local/lib/python2.7/dist-packages/ds4drv/uinput.py", line 268, in create_device
    product=layout.product, version=layout.version)
  File "/usr/local/lib/python2.7/dist-packages/evdev/uinput.py", line 107, in __init__
    _uinput.create(self.fd, name, vendor, product, version, bustype, absinfo)
IOError: [Errno 22] Invalid argument
```

I then downloaded/installed 0.5.1 from github and it works now. Suggest you push 0.5.1 to pypi.

Thanks!!!

Ape, 2016-07-29:
But isn't 0.5.1 in pypi? https://pypi.python.org/pypi/ds4drv

walchko, 2016-07-29:
ok, so i looked and it says it is, however, when I downloaded it yesterday it didn’t work. When I found the issue and pulled 0.5.1 from github it worked. I am confused.

Ape, 2016-07-29:
Ok, it needs to be tested if `pip install` still downloads 0.5.0 in some situations or systems.

alextricity25, 2016-08-02:
I'm running Ubuntu 16.04, pip version 8.1.2, kernel 4.4.0-31-generic.
Running "sudo pip install ds4drv" installed version 0.5.0 for me. I've pasted my output.
```
miguel@miguel-All-Series:~$ sudo pip install ds4drv
The directory '/home/miguel/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/miguel/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting ds4drv
Requirement already satisfied (use --upgrade to upgrade): pyudev>=0.16 in ./.local/lib/python2.7/site-packages (from ds4drv)
Requirement already satisfied (use --upgrade to upgrade): evdev>=0.3.0 in ./.local/lib/python2.7/site-packages (from ds4drv)
Requirement already satisfied (use --upgrade to upgrade): six in ./.local/lib/python2.7/site-packages (from pyudev>=0.16->ds4drv)
Installing collected packages: ds4drv
Successfully installed ds4drv-0.5.0
```

However, running a "pip search ds4drv" does show that 0.5.1 is available:
```
miguel@miguel-All-Series:~$ pip search ds4drv
ds4drv (0.5.1)  - A Sony DualShock 4 userspace driver for Linux
  INSTALLED: 0.5.0
  LATEST:    0.5.1
```

Trying to upgrade doesn't result in anything:
```
miguel@miguel-All-Series:~$ sudo pip install ds4drv --upgrade
The directory '/home/miguel/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/miguel/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already up-to-date: ds4drv in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: pyudev>=0.16 in ./.local/lib/python2.7/site-packages (from ds4drv)
Requirement already up-to-date: evdev>=0.3.0 in ./.local/lib/python2.7/site-packages (from ds4drv)
Requirement already up-to-date: six in ./.local/lib/python2.7/site-packages (from pyudev>=0.16->ds4drv)
miguel@miguel-All-Series:~$ ds4drv --version
ds4drv 0.5.0
```

Sounds like others are having the same issue with installing 0.5.1 from pypi.
Let me know if you need any more information.

prettyv, 2016-08-06:
Testing on current Arch Linux in a fresh virtualenv pip seems to do the right thing:
```
$ pip install ds4drv -v
Collecting ds4drv
  1 location(s) to search for versions of ds4drv:
  * https://pypi.python.org/simple/ds4drv/
  Getting page https://pypi.python.org/simple/ds4drv/
  Looking up "https://pypi.python.org/simple/ds4drv/" in the cache
  Current age based on date: 690
  Freshness lifetime from max-age: 600
  Freshness lifetime from request max-age: 600
  The cached response is "stale" with no etag, purging
  Starting new HTTPS connection (1): pypi.python.org
  "GET /simple/ds4drv/ HTTP/1.1" 200 953
  Updating cache with response from "https://pypi.python.org/simple/ds4drv/"
  Caching b/c date exists and max-age > 0
  Analyzing links from page https://pypi.python.org/simple/ds4drv/
    Found link https://pypi.python.org/packages/05/62/68d78b88dfec8917e678ae7d516a603af169e83d58b4572006cc084ea0ce/ds4drv-0.5.1-py3-none-any.whl#md5=1a06074a168bc3633c74814a03053d3a (from https://pypi.python.org/simple/ds4drv/), version: 0.5.1
    Found link https://pypi.python.org/packages/0e/1f/07152d3d48d8941d737308ade4e2b4bd682dda8c50a05aa92b0e8e3aa36c/ds4drv-0.2.0.tar.gz#md5=dab54e414e4a96ad7814e1555e97e52a (from https://pypi.python.org/simple/ds4drv/), version: 0.2.0
    Found link https://pypi.python.org/packages/10/45/fe8bc3c030249325e4c408618563f7b689eba3be0d34dd1499c755118502/ds4drv-0.4.3.tar.gz#md5=3be7a154d94372613f6daa46a3bfccb8 (from https://pypi.python.org/simple/ds4drv/), version: 0.4.3
    Found link https://pypi.python.org/packages/44/33/2317e945fb55923694a3c27b49c57e36f407e3e58e0da159f833b1bb7d3b/ds4drv-0.3.0.tar.gz#md5=5e79c90a38646fbe9f29430927639665 (from https://pypi.python.org/simple/ds4drv/), version: 0.3.0
    Found link https://pypi.python.org/packages/4b/39/3ab5cf7e53ffa3ace741ce97aed5150c18bfc08747366655253e3a1c3ac0/ds4drv-0.1.0.tar.gz#md5=ee6fe7917eec642270ba29483e61cd57 (from https://pypi.python.org/simple/ds4drv/), version: 0.1.0
    Found link https://pypi.python.org/packages/57/b2/1cd3c405dbbeda634666b3ebda5568f931cc054d8497b3b14a86100d8050/ds4drv-0.2.1.tar.gz#md5=361483cddfbbc7ed259ea31250ddf061 (from https://pypi.python.org/simple/ds4drv/), version: 0.2.1
    Found link https://pypi.python.org/packages/7e/5a/3c6b6d2470e3bf4026de01c8d9eda4ea3e743e9245402e1effd55ffc5791/ds4drv-0.4.1.tar.gz#md5=29b4fe5a390a9ee06d32d5a0c657822c (from https://pypi.python.org/simple/ds4drv/), version: 0.4.1
    Found link https://pypi.python.org/packages/cf/98/47f1a94e72b4d391be75db5280f451897ee4b2f3de1009f0229ee22a6bee/ds4drv-0.4.2.tar.gz#md5=8c2d210023f27872ba28e04f1ecb5cc4 (from https://pypi.python.org/simple/ds4drv/), version: 0.4.2
    Found link https://pypi.python.org/packages/d4/48/7bb6d283b9b8390dc095d135d9ff84fa359d0e7dbd62d1fa85361713bb47/ds4drv-0.4.0.tar.gz#md5=a0f1f1c6ba2d9214c7a133439e3777ab (from https://pypi.python.org/simple/ds4drv/), version: 0.4.0
    Found link https://pypi.python.org/packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz#md5=705d48f45634f9dbd3f40c8876fb2c96 (from https://pypi.python.org/simple/ds4drv/), version: 0.5.0
    Found link https://pypi.python.org/packages/f5/04/f4f246ac374d59cf5d03a631854b67140de0dd6b3a77cc02f8b8c4886589/ds4drv-0.1.1.tar.gz#md5=a3e21aeaea79b1faed0616d1ca015eed (from https://pypi.python.org/simple/ds4drv/), version: 0.1.1
  Using version 0.5.1 (newest of versions: 0.1.0, 0.1.1, 0.2.0, 0.2.1, 0.3.0, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.5.0, 0.5.1)
  Looking up "https://pypi.python.org/packages/05/62/68d78b88dfec8917e678ae7d516a603af169e83d58b4572006cc084ea0ce/ds4drv-0.5.1-py3-none-any.whl" in the cache
  Current age based on date: 690
  Freshness lifetime from max-age: 31557600
  The response is "fresh", returning cached response
  31557600 > 690
  Using cached ds4drv-0.5.1-py3-none-any.whl
  Downloading from URL https://pypi.python.org/packages/05/62/68d78b88dfec8917e678ae7d516a603af169e83d58b4572006cc084ea0ce/ds4drv-0.5.1-py3-none-any.whl#md5=1a06074a168bc3633c74814a03053d3a (from https://pypi.python.org/simple/ds4drv/)
[...]
Successfully installed ds4drv-0.5.1 evdev-0.6.1 pyudev-0.21.0 six-1.10.0
```

Maybe it would help if anyone hitting the problem posts pip's verbose output (-v)?

alextricity25, 2016-08-15:
This is what i got when running ``sudo pip install ds4drv -v``:
```
Collecting ds4drv
  1 location(s) to search for versions of ds4drv:
  * https://pypi.python.org/simple/ds4drv/
  Getting page https://pypi.python.org/simple/ds4drv/
  Looking up "https://pypi.python.org/simple/ds4drv/" in the cache
  No cache entry available
  Starting new HTTPS connection (1): pypi.python.org
  "GET /simple/ds4drv/ HTTP/1.1" 200 953
  Updating cache with response from "https://pypi.python.org/simple/ds4drv/"
  Caching b/c date exists and max-age > 0
  Analyzing links from page https://pypi.python.org/simple/ds4drv/
    Skipping link https://pypi.python.org/packages/05/62/68d78b88dfec8917e678ae7d516a603af169e83d58b4572006cc084ea0ce/ds4drv-0.5.1-py3-none-any.whl#md5=1a06074a168bc3633c74814a03053d3a (from https://pypi.python.org/simple/ds4drv/); it is not compatible with this Python
    Found link https://pypi.python.org/packages/0e/1f/07152d3d48d8941d737308ade4e2b4bd682dda8c50a05aa92b0e8e3aa36c/ds4drv-0.2.0.tar.gz#md5=dab54e414e4a96ad7814e1555e97e52a (from https://pypi.python.org/simple/ds4drv/), version: 0.2.0
    Found link https://pypi.python.org/packages/10/45/fe8bc3c030249325e4c408618563f7b689eba3be0d34dd1499c755118502/ds4drv-0.4.3.tar.gz#md5=3be7a154d94372613f6daa46a3bfccb8 (from https://pypi.python.org/simple/ds4drv/), version: 0.4.3
    Found link https://pypi.python.org/packages/44/33/2317e945fb55923694a3c27b49c57e36f407e3e58e0da159f833b1bb7d3b/ds4drv-0.3.0.tar.gz#md5=5e79c90a38646fbe9f29430927639665 (from https://pypi.python.org/simple/ds4drv/), version: 0.3.0
    Found link https://pypi.python.org/packages/4b/39/3ab5cf7e53ffa3ace741ce97aed5150c18bfc08747366655253e3a1c3ac0/ds4drv-0.1.0.tar.gz#md5=ee6fe7917eec642270ba29483e61cd57 (from https://pypi.python.org/simple/ds4drv/), version: 0.1.0
    Found link https://pypi.python.org/packages/57/b2/1cd3c405dbbeda634666b3ebda5568f931cc054d8497b3b14a86100d8050/ds4drv-0.2.1.tar.gz#md5=361483cddfbbc7ed259ea31250ddf061 (from https://pypi.python.org/simple/ds4drv/), version: 0.2.1
    Found link https://pypi.python.org/packages/7e/5a/3c6b6d2470e3bf4026de01c8d9eda4ea3e743e9245402e1effd55ffc5791/ds4drv-0.4.1.tar.gz#md5=29b4fe5a390a9ee06d32d5a0c657822c (from https://pypi.python.org/simple/ds4drv/), version: 0.4.1
    Found link https://pypi.python.org/packages/cf/98/47f1a94e72b4d391be75db5280f451897ee4b2f3de1009f0229ee22a6bee/ds4drv-0.4.2.tar.gz#md5=8c2d210023f27872ba28e04f1ecb5cc4 (from https://pypi.python.org/simple/ds4drv/), version: 0.4.2
    Found link https://pypi.python.org/packages/d4/48/7bb6d283b9b8390dc095d135d9ff84fa359d0e7dbd62d1fa85361713bb47/ds4drv-0.4.0.tar.gz#md5=a0f1f1c6ba2d9214c7a133439e3777ab (from https://pypi.python.org/simple/ds4drv/), version: 0.4.0
    Found link https://pypi.python.org/packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz#md5=705d48f45634f9dbd3f40c8876fb2c96 (from https://pypi.python.org/simple/ds4drv/), version: 0.5.0
    Found link https://pypi.python.org/packages/f5/04/f4f246ac374d59cf5d03a631854b67140de0dd6b3a77cc02f8b8c4886589/ds4drv-0.1.1.tar.gz#md5=a3e21aeaea79b1faed0616d1ca015eed (from https://pypi.python.org/simple/ds4drv/), version: 0.1.1
  Using version 0.5.0 (newest of versions: 0.1.0, 0.1.1, 0.2.0, 0.2.1, 0.3.0, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.5.0)
  Looking up "https://pypi.python.org/packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz" in the cache
  No cache entry available
  "GET /packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz HTTP/1.1" 200 25807
  Downloading ds4drv-0.5.0.tar.gz
  Downloading from URL https://pypi.python.org/packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz#md5=705d48f45634f9dbd3f40c8876fb2c96 (from https://pypi.python.org/simple/ds4drv/)
  Updating cache with response from "https://pypi.python.org/packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz"
  Caching due to etag
  Running setup.py (path:/tmp/pip-build-X1S2Jq/ds4drv/setup.py) egg_info for package ds4drv
    Running command python setup.py egg_info
    running egg_info
    creating pip-egg-info/ds4drv.egg-info
    writing requirements to pip-egg-info/ds4drv.egg-info/requires.txt
    writing pip-egg-info/ds4drv.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/ds4drv.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/ds4drv.egg-info/dependency_links.txt
    writing entry points to pip-egg-info/ds4drv.egg-info/entry_points.txt
    writing manifest file 'pip-egg-info/ds4drv.egg-info/SOURCES.txt'
    warning: manifest_maker: standard file '-c' not found

    reading manifest file 'pip-egg-info/ds4drv.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'pip-egg-info/ds4drv.egg-info/SOURCES.txt'
  Source in /tmp/pip-build-X1S2Jq/ds4drv has version 0.5.0, which satisfies requirement ds4drv from https://pypi.python.org/packages/f3/91/5811c4f9d1ece2b74c28cc1f702be76cc2bac1c2b5e5de8465de438ada26/ds4drv-0.5.0.tar.gz#md5=705d48f45634f9dbd3f40c8876fb2c96
Requirement already satisfied (use --upgrade to upgrade): evdev>=0.3.0 in /home/miguel/.local/lib/python2.7/site-packages (from ds4drv)
Requirement already satisfied (use --upgrade to upgrade): pyudev>=0.16 in /home/miguel/.local/lib/python2.7/site-packages (from ds4drv)
[... cut ...]
    reading manifest file 'ds4drv.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'ds4drv.egg-info/SOURCES.txt'
    Copying ds4drv.egg-info to /usr/local/lib/python2.7/dist-packages/ds4drv-0.5.0.egg-info
    running install_scripts
    Installing ds4drv script to /usr/local/bin
    writing list of installed files to '/tmp/pip-KiOh93-record/install-record.txt'
done
  Removing source in /tmp/pip-build-X1S2Jq/ds4drv
Successfully installed ds4drv-0.5.0
Cleaning up...
```

prettyv, 2016-08-15:
Ah, I see.
@Ape: You seem to have uploaded a py3-only wheel for 0.5.1, as opposed to universal tarballs for the previous releases (as can be seen on https://pypi.python.org/simple/ds4drv/), which is why it works in my py3-virtualenv, and not on installations using py2 by default. So a repackaging is needed.

Ape, 2016-08-15:
Merging #58 should solve the issue, right?

Ape, 2016-08-15:
I uploaded a new package for 0.5.1. Please confirm that it works now. If so, I will merge #58 so that I don't have to remember to add `--universal` in the future.

prettyv, 2016-08-16:
Yes, configuring setuptools to build universal wheels is what's needed, so that pull request can be merged. 0.5.1 installs successfully now on both py3 and py2, although on py3 it still uses the previously built py3-only wheel. Not sure if that warrants a version bump just for a packaging fix, but it seems to not strictly be needed in this case.

Ape, 2016-08-17:
I think this is fine like this. I hope I can release 0.6.0 soon.

closed 2016-08-17 by Ape

graingert, 2016-08-17:
There's no such thing as a universal tar ball. Do you mean an sdist?

graingert, 2016-08-17:
`python3 setup.py sdist bdist_wheel register upload`

To push the sdist and the wheels

Ape, 2016-08-17:
I meant the universal wheel. Publishing only the sdist might be fine, too. I will push both sdist and the universal wheel for any future releases.

commit bbec9306 (2016-08-17): Merge pull request #58 from graingert/patch-1

Support universal wheels

Fixes #95.