chrippa/ds4drv #97: --emulate-xpad-wireless not working - no games are responding to controller

Enigma0, 2016-08-28:
I'm launching with `ds4drv --trackpad-mouse --emulate-xpad-wireless --hidraw` and it is recognized in Steam and can launch games but no game is responding to any controller input.

Running Ubuntu GNOME 16.04 with latest steam and installed `ds4drv` via `pip install`.

Ape, 2016-08-29:
Does it work if you do not run ds4drv at all, but instead just launch the game from Steam Big Picture Mode (you said that you use Steam)? It's not a solution to make ds4drv work, but it might be a workaround for you and perhaps give us more information about the situation.

Enigma0, 2016-08-29:
Yup I use Steam. I tried initially just using the system's Bluetooth and I it didn't work at all (not even Big Picture).

Enigma0, 2016-08-31:
Also similar to these bugs possibly:

- https://github.com/chrippa/ds4drv/issues/52
- https://github.com/chrippa/ds4drv/issues/45

Neither seem to have a solution.