chrippa/ds4drv #100: default mapping names and ordering wrong for axes

gomisensei, 2016-09-25:
so, i noticed that the axes order seems to be wierd, not sure if this is intended, but the order of the axes in jstest has the left and right analog triggers (named Z and Rx) default mapped between left stick vertical and horizontal (named Z and Rz).  The last 3 axes are unnamed, they are the acceleration sensors, X Y Z,  

Also, the 2 analog triggers, (which on the controller are labelled L2 and R2) go from - when released, to 0 when half pressed, to + at full, while also registering as buttons.

gomisensei, 2016-09-25:
this information is based on jstest... the --dump-reports option seems to show everything correctly...

timrose-, 2016-12-10:
I can also confirm this issue.

uinput.py contains references to the triggers under the axes mapping (lines 73 and 74). Tried removing these lines but appear to of broke it further...