chrippa/ds4drv #104: Hidraw USB mode doesn't work with DS4 Slim

Alexay, 2016-10-28:
Trying to get the Dual Shock 4 Slim controller to work through USB, because Bluetooth mode has too much input lag (but it works).

Ubuntu 16.04 x64,
Installed stable ds4drv through `pip install ds4rv`,
Ran `sudo ds4rv --hidraw`
The output is:
```
[info][controller 1] Created devices /dev/input/js1 (joystick) /dev/input/event16 (evdev) 
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
```

Every time I disconnect and reconnect the controller USB cable it does a scan for devices but doesn't detect anything.

My `lsusb` sees the controller as:
`Bus 003 Device 008: ID 054c:09cc Sony Corp. `

When connecting the controller through USB `dmesg` outputs the following:
```
[583905.131514] usb 3-10: new full-speed USB device number 9 using xhci_hcd
[583905.260575] usb 3-10: New USB device found, idVendor=054c, idProduct=09cc
[583905.260577] usb 3-10: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[583905.260578] usb 3-10: Product: Wireless Controller
[583905.260579] usb 3-10: Manufacturer: Sony Interactive Entertainment
[583905.260649] usb 3-10: ep 0x84 - rounding interval to 32 microframes, ep desc says 40 microframes
[583905.260651] usb 3-10: ep 0x3 - rounding interval to 32 microframes, ep desc says 40 microframes
[583905.271098] input: Sony Interactive Entertainment Wireless Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.3/0003:054C:09CC.0049/input/input125
[583905.271196] hid-generic 0003:054C:09CC.0049: input,hidraw1: USB HID v1.11 Gamepad [Sony Interactive Entertainment Wireless Controller] on usb-0000:00:14.0-10/input3
```

Alexay, 2016-10-28:
Looks like the problem is that the device name is different for the DS4 Slim, I've managed to get ds4drv to work by changing one line in `/backends/hidraw.py`:

Before:
```python
HID_DEVICES = {
    "Sony Computer Entertainment Wireless Controller": HidrawUSBDS4Device,
    "Wireless Controller": HidrawBluetoothDS4Device,
}
```

After:
```python
HID_DEVICES = {
    "Sony Interactive Entertainment Wireless Controller": HidrawUSBDS4Device,
    "Sony Computer Entertainment Wireless Controller": HidrawUSBDS4Device,
    "Wireless Controller": HidrawBluetoothDS4Device,
}
```

I'll create a pull request.

closed 2016-10-28 by Ape

stevenqubrossi, 2017-01-23:
Hi guys, sorry I'm new to the world of retro pie. How can I find the directory for /backends/hidraw.py? Like where is this located so I can edit the above information? I am using PuTTY to access the backend, but I can't seem to find it anywhere....

schrottulk, 2018-01-09:
Thanks that worked for me! I found the directory in: /usr/local/lib/python3.5/dist-packages/ds4drv/backends/

yiufung, 2019-11-11:
@Alexay This saves my day! Wonder why it's not merged yet.

jackbluecx, 2022-08-27:
didn't work here for me.  I did everything as shown above, does anyone have any other tips.

bernardoaraujor, 2022-11-12:
works for me, thanks so much!

Cebastion, 2025-01-08:
It doesn't work. When I edited hidraw.py and entered sudo ds4drv --hidraw/ ds4drv --hidraw. It returned 

`
sudo ds4drv --hidraw
[info][controller 1] Created devices /dev/input/js1 (joystick) /dev/input/event25 (evdev) 
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 33, in <module>
    sys.exit(load_entry_point('ds4drv==0.5.1', 'console_scripts', 'ds4drv')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyudev-0.24.3-py3.12.egg/ds4drv/__main__.py", line 154, in main
    for device in backend.devices:
  File "/usr/local/lib/python3.12/dist-packages/pyudev-0.24.3-py3.12.egg/ds4drv/backends/hidraw.py", line 174, in devices
    yield cls(name=device_name,
          ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyudev-0.24.3-py3.12.egg/ds4drv/backends/hidraw.py", line 34, in __init__
    super(HidrawDS4Device, self).__init__(name, addr, type)
  File "/usr/local/lib/python3.12/dist-packages/pyudev-0.24.3-py3.12.egg/ds4drv/device.py", line 84, in __init__
    self.set_operational()
  File "/usr/local/lib/python3.12/dist-packages/pyudev-0.24.3-py3.12.egg/ds4drv/backends/hidraw.py", line 100, in set_operational
    addr = self.read_feature_report(0x81, 6)[1:]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/pyudev-0.24.3-py3.12.egg/ds4drv/backends/hidraw.py", line 63, in read_feature_report
    return fcntl.ioctl(self.fd, op, bytes(buf))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BrokenPipeError: [Errno 32] Broken pipe
`

commit 49503343 (2016-10-28): Fixed #104