chrippa/ds4drv #107: Device created before anything is connected?

zacwitte, 2016-11-04:
It looks like when ds4drv is started it automatically creates a Sony Computer Entertainment Wireless Controller device, which evdev thinks is real even though no controller is connected via USB or Bluetooth. Once a controller is connected, it starts sending events and when it's disconnected it stops sending events, but the input device remains. My application needs to be certain when a controller is connected and when it's not. Is this possible with ds4drv?

Ape, 2016-11-06:
I think this behavior is intentional to help with games that do not handle hot plugging very well. The ds4drv behavior is debatable, but generally you shouldn't assume that every connected controller is something that the player wants to use right now on this game. People often have extra controllers connected at all times and some of them might be special things like steering wheels.

You should make a splash screen to your game saying something like "Press A to start" and detect the controller on that. Or if it's multiplayer, then a screen where you can press A on each controller that you would like to use.

zacwitte, 2016-11-06:
My use case is a little bit different. I'm using the ds4 as a remote
control for a robot. I can implement some kind of press A to start as a
hack to detect if a controller is connected but it's actually really
important that I detect if the controller is disconnected. I need to
immediately stop the robot. If I wanted to modify the source code so that
the input device were created only when the controller is connected and
removed immediately when disconnected so I can handle the error, how hard
would that be? Can you point me in the right direction?

Ape, 2016-11-07:
It's possible to modify the code so that the ds4drv joystick device is added and removed dynamically, but I think it would be easier if you just detect whether the actual ds4 joystick device is present. Perhaps you could just poll `/proc/bus/input/devices`.

zacwitte, 2016-11-08:
That's a good point - finding the sysfs of the real device from `/proc/bus/input/devices` and monitoring it to see when the path disappears seems to be a pretty reliable way to detect when the controller is disconnected. I can similarly monitor `/proc/bus/input/devices` to see when a controller is connected.

It doesn't quite solve the situation of multiple controllers, but that's an edge case I don't need to worry about right now.

closed 2016-11-08 by zacwitte

leosmeira, 2017-07-16:
It would be great if inputs were created dynamically.
Starting ds4drv at boot, it will always use js0. If we do not use the PS4 controller, some emulators will reserve the first player for js0.
I think the best way is to run the dynamic creation as it is done in Sixad PS3.