chrippa/ds4drv #109: LED light is very dim

zacwitte, 2016-11-14:
I'm not sure if this is an issue with ds4drv, but the LED light on my controller is very dim once it's connected. While it's connecting or in pairing mode it flashes at a normal brightness, but when it's connected I need to cup my hand over the light to see it. Is this something anyone else is experiencing? I've tried setting the led color to ff0000 and 0000ff.

Ape, 2016-11-15:
Does the color change to red and blue when you set the color?

zacwitte, 2016-11-16:
The color doesn't change - it's always a dim red. I've tried it with hidraw enabled and disabled.

Alkaris, 2018-09-13:
Sounds like it's not reading from your conf file. Make sure there's a `ds4drv.conf` file in `~/.config` directory, if not get it from here https://github.com/chrippa/ds4drv/blob/master/ds4drv.conf

it seems very normal behaviour on connecting the controller to the computer to have a dimmed LED if there is no config file present I've found, except the default color is usually blue, not red.