chrippa/ds4drv #122: jstest does not report keypresses on device created with ds4drv

buha, 2017-03-04:
$ uname -r
    4.9.13-1-MANJARO

    $ cat /etc/udev/rules.d/50-ds4drv.rules
    KERNEL=="uinput", MODE="0666"
    KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="054c", ATTRS{idProduct}=="05c4", MODE="0666"
    KERNEL=="hidraw*", SUBSYSTEM=="hidraw", KERNELS=="0005:054C:05C4.*", MODE="0666"
    KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="054c", ATTRS{idProduct}=="09cc", MODE="0666"
    KERNEL=="hidraw*", SUBSYSTEM=="hidraw", KERNELS=="0005:054C:09CC.*", MODE="0666"

    # By plugging the DS4 controller, I get this
    $ ls -l /dev/input/by-id/ | grep Controller
    lrwxrwxrwx 1 root root  9 Mar  4 11:59 usb-Sony_Interactive_Entertainment_Wireless_Controller-if03-event-joystick -> ../event1
    lrwxrwxrwx 1 root root  6 Mar  4 11:59 usb-Sony_Interactive_Entertainment_Wireless_Controller-if03-joystick -> ../js0

At this point I launch `ds4drv` (tried with `--hidraw` as well) and it creates another device at `/dev/input/js1`. The problem is that `jstest / jstest-gtk` dont pick any keypresses on the `.../js1`, only the original `.../js0' shows action.

Any idea how to get this working?

randokando, 2017-03-04:
Is this similar to [my issue](https://github.com/chrippa/ds4drv/issues/119)?

buha, 2017-03-20:
Hard for me to tell because I don't have this retropie thing. In any case, your `ghost controller` term applies well to what I experience...

In the meantime I played some games over bluetooth, works fine. But over USB it's still the same as above.

randokando, 2017-03-20:
@buha 

Guh —  
I'm jealous.  
No bluetooth capability for me.

buha, 2017-03-28:
Make sure you have at least a bluetooth 2.1 adapter connected to your Pi
Make sure pairing is ok
I simply run `ds4drv` and it works for me. Maybe I'm lucky, yeah