chrippa/ds4drv #124: Clean exit on SIGINT?

joseph-norman, 2017-03-12:
Hi guys,

Love the work you've done with this - works perfectly on my Ubuntu machine! Would you guys be interested in a PR that exited cleanly (i.e, without a stack trace) on a SIGINT in an active terminal?

Cheers!

Ape, 2017-03-13:
This should already work correctly. Here's an example output for me:

```
[info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event19 (evdev) 
[info][controller 1] Connected to Bluetooth Controller (1C:66:6D:1A:6F:A0 hidraw2)
[info][hidraw] Scanning for devices
[info][controller 1] Battery: Fully charged
^C[info][controller 1] Disconnected
[info][controller 1] Cleaning up...
```

Ape, 2017-03-13:
Does this only occur rarely for you? Can you try if Python 3 changes anything?

joseph-norman, 2017-03-13:
I'll give Python 3 a try out - it occurs every time I close

Ciruxx, 2019-07-27:
Same here with Arch Linux!