chrippa/ds4drv #125: Bad mappings in Dark Souls on Wine via hidraw

hertz4, 2017-03-18:
I'm trying to play Dark Souls 1 on Steam via Wine, and --hidraw --emulate-xboxdrv puts the buttons in pretty random places that don't exactly make the game playable. --emulate-xpad doesn't even work in the first place, because something silly is mapped to the right stick and the camera just spins. Here's what it's doing:

Good mappings:
- L1, R2 - LB, RB
- Lstick, Rstick - Lstick, Rstick

Bad mappings:
- Circle - Y
- Triangle - B
- Cross - X
- Square, L2, R2 - Nothing

Now, when I don't use ds4drv, the mappings are identical to Windows, so I can assume that the game is reading them the same way. I can elaborate on how it maps things there if that helps. I also don't know whether this is an issue in other games via Wine, but I do know that Souls 1 is known for funny behavior with gamepads.

hertz4, 2017-03-20:
This works for some reason. Left stick button and PS button might need to be swapped (souls doesn't use it), but at least I can play the game now.

```python
create_mapping(
    "souls", "Make Dark Souls work with DS4",
    # Bus type, vendor, product, version                                                                                                                                                      
    0,          0,      0,       0,
    # Axes                                                                                                                                                                                    
    {
	"ABS_X":     "left_analog_x",
	"ABS_Y":     "left_analog_y",
	"ABS_RX":    "right_analog_x",
	"ABS_RY":    "right_analog_y",
    },
    # Axes settings                                                                                                                                                                           
    {},
    #Buttons                                                                                                                                                                                  
    {
        "BTN_SELECT": "button_ps",
	"BTN_THUMBL": "button_options",
	"BTN_MODE":   "button_share",
	"BTN_X":      "button_cross",
	"BTN_Y":      "button_circle",
	"BTN_A":      "button_square",
	"BTN_B":      "button_triangle",
	"BTN_TL":     "button_l1",
	"BTN_TR":     "button_r1",
	"BTN_TL2":    "button_l2",
	"BTN_TR2":    "button_r2",
	"BTN_THUMBR": "button_l3",
	"BTN_START":  "button_r3"
    },
    # Hats                                                                                                                                                                                    
    {
	"ABS_HAT0X": ("dpad_left", "dpad_right"),
	"ABS_HAT0Y": ("dpad_up", "dpad_down")
    }
)
```

Soofe, 2017-06-17:
Hello,

Have you tried with Dark Souls II ? It doesn't seem to recognize the controller.

Ryochan7, 2017-08-05:
Has anyone ever tried Dark Souls using an actual Xbox 360 controller? Similar problems come up when using a real Xbox 360 controller while trying to play Dark Souls on Wine; similar problems also come up when trying to use a DS4 to play Dark Souls on Windows without using software to make a virtual Xbox 360 controller. This particular problem is not just an issue with ds4drv.

The spinning camera is due to ABS_Z being mapped to L2 and ABS_RZ being mapped to R2. Dark Souls in DirectInput mode apparently uses those axes as well as ABS_RX and ABS_RY for camera rotation. Dark Souls interprets the minimum trigger values as the max negative value for an axis as opposed to 0 so the neutral point ends up being when the triggers are at half pull; evtest seems to show that proper scaling is not done in ds4drv.

Buttons are swapped around but that problem exists when using a real Xbox 360 controller as well.