chrippa/ds4drv #133: ds4drv joystick device is added and removed dynamically.

leosmeira, 2017-07-16:
It would be great if inputs were created dynamically.
Starting ds4drv at boot, it will always use js0. If we do not use the PS4 controller, some emulators will reserve the first player for js0.
I think the best way is to run the dynamic creation as it is done in Sixad PS3.

Ape, 2017-07-18:
I would be ok with merging some `--dynamic` option that makes ds4drv do this.

Alkaris, 2018-09-13:
I should add that if you've got a ds4drv daemon already running in the background, and you plug in the controller and reads from a config file, it'll move the controller to `/dev/js1` because it thinks `/dev/js0` is already taken by the daemon with another controller, instead of hooking into the already existing daemon, so you end up what looks like multiple controllers connected to the system when there is only 1 connected. This is also further shown when you open up `jstest` GUI app, it'll show as if two controllers are connected at once, this can somewhat cause issues when reading the `ds4drv.conf` file and some readings of the configurations not working properly.