chrippa/ds4drv #134: no led when hidraw enabled

Scrumplex, 2017-07-20:
If using hidraw mode, the led color is not changing. It can be really confusing when changing profiles, without any feedback.
ds4drv version 0.5.1 from pip is in use.

fullmetal1, 2017-08-03:
This also happens to me. ds4drv 0.5.1 from the aur

b2ag, 2017-09-09:
Same here. Using Arch Linux packge ds4drv-git from AUR.
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ds4drv-git

Strange thing: LED change works for me using hidraw when connecting via USB.

Scrumplex, 2017-10-22:
I am currently looking through the code for a fix. As far as I know the packet structure is correct. I just tried to capture bluetooth0 with Wireshark, when I only saw incoming packets. So ds4drv's outgoing report packets are not being sent. Investigating.

indeedwatson, 2018-05-11:
Same issue, also in Arch.

Is this project still alive?

PatrickHussey, 2018-08-18:
#154 has this fixed

taleden, 2019-10-30:
#169 also fixes this, in (AFAICT) a slightly better way (no added zlib dependency, and works on python3)