chrippa/ds4drv #144: Touchpad not working. Needs to be disabled in order to work

Scrumplex, 2018-01-31:
On my setup, if I try to connect my DS4 to my PC (hidraw) and set anything for touchpad setting in ds4drv it is just ignored, because it is overridden by Linux kernel. If I disable the xinput device created by the kernel it works. Any ideas, how I can disable it permanently?

Sur3, 2018-02-11:
@Scrumplex what do you mean with "xinput device created by the kernel" is there a kernel driver for the ds4 touchpad now?

Sur3, 2018-02-12:
Ok I take that back found the kernel module for touchpad now:  
"input: Sony Interactive Entertainment Wireless Controller Touchpad as /devices/pci.../.../input/input23"
But it it not shown on xinput, so probably a xinput driver is missing for me.

Sur3, 2018-02-12:
@Scrumplex can you look up which input driver xinput is using for the kernel driver? Synaptics, libinput or evdev or something totally different? Think you should find that in /var/log/Xorg.0.log or are you using wayland?

Sur3, 2018-02-12:
Ok I got the touchpad working with evdev as input driver. Is there any reason to use ds4drv for the touchpad if the native kernel driver is working?