chrippa/ds4drv #146: Control DS4 events and inputs

Alkaris, 2018-04-16:
Some of the configurations options for ds4drv doesn't work when you want to switch profiles to disable or enable certain features on the PS4 controller. The Touchpad input for one, even if you set in the `ds4drv.conf` to set trackpad config  `trackpad-mouse = false` it doesn't do that. Mouse input is still usable and controls the mouse cursor on screen when it shouldn't. If I run `xinput disable {id}` for the PS4 controller only then can it be disabled and still have the rest of the normal functions of the PS4 controller.

Another thing is not being able to override the gyro sensors in the PS4 controller, which makes it impossible remap buttons because the axis controls are taking over the input changes. Even if the controller is sitting perfectly flat, the gyro sensors are still acting crazy and all over the place. Also because the way ds4drv acts as a controller input device, it ends up placing the controller into 2 other /dev/input's so it looks like I have 3 controllers connected when it's only 1 connected `ds4drv` shows up as `/dev/input/js0` and the other two show up as `/dev/input/js1` and `/dev/input/js2`

[image]

`/dev/input/js1` appears for Motion Sensors, and `/dev/input/js2` appears for Touchpad.

[image]

Need to have a way to disable these functions when plugged in and ds4drv is in use as the primary driver to control them, or create some rule to not be able to make other inputs, but that may make things difficult when trying to plugin more controllers into one machine.

turtleDev, 2018-10-01:
what tool are you using to visualise joystick inputs?

Alkaris, 2018-10-01:
@turtleDev I'm using jstest.