chrippa/ds4drv #148: module 'socket' has no attribute 'AF_BLUETOOTH'

SebastianPfliegel, 2018-04-26:
This is the output which gets triggered as soon as I try to pair the device:

```
[info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event21 (evdev) 
[info][bluetooth] Scanning for devices
[info][bluetooth] Found device A4:53:85:38:DB:02
Traceback (most recent call last):
  File "/usr/bin/ds4drv", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python3.6/site-packages/ds4drv/__main__.py", line 154, in main
    for device in backend.devices:
  File "/usr/lib/python3.6/site-packages/ds4drv/backends/bluetooth.py", line 132, in devices
    device = self.find_device()
  File "/usr/lib/python3.6/site-packages/ds4drv/backends/bluetooth.py", line 121, in find_device
    return BluetoothDS4Device.connect(bdaddr)
  File "/usr/lib/python3.6/site-packages/ds4drv/backends/bluetooth.py", line 23, in connect
    ctl_socket = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_SEQPACKET,
AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'
```

Any idea on how to fix this? Thanks in advance. Btw, using openSUSE Tumbleweed.

roadh0use, 2018-08-17:
must be an opensuse problem
same error message here
since this is from 4 months ago without a response im assuming there is no fix so far is there?

JackKiefer, 2019-03-30:
I'm having the same problem on Ubuntu 18.04

JackKiefer, 2019-04-01:
Update:
I was able to fix the problem by just cloning the repo and installing from source. Seems like it might just be the version of ds4drv installed through pip that is the issue.

smoothdvd, 2019-10-16:
@JackMiranda  same here with Ubuntu 18.04

cslev, 2020-09-15:
Your problem is that the python version you are using does have proper Bluetooth support! I had the same problem when installed python from source and did not have libbluetooth-dev package installed on my system! So, install libbluetooth-dev (or something similar in other than Debian), and reinstall python. The easiest way to check it is working is to fireup a python console from terminal and try this:
```
from socket import AF_BLUETOOTH
```

Y0ngg4n, 2021-04-24:
Having the same problem on arch with the newest bluez-utils-compat release 
```
[info][bluetooth] Scanning for devices
[info][bluetooth] Found device A4:AE:12:AD:5B:C6
Traceback (most recent call last):
  File "/usr/bin/ds4drv", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/ds4drv/__main__.py", line 154, in main
    for device in backend.devices:
  File "/usr/lib/python3.9/site-packages/ds4drv/backends/bluetooth.py", line 132, in devices
    device = self.find_device()
  File "/usr/lib/python3.9/site-packages/ds4drv/backends/bluetooth.py", line 121, in find_device
    return BluetoothDS4Device.connect(bdaddr)
  File "/usr/lib/python3.9/site-packages/ds4drv/backends/bluetooth.py", line 23, in connect
    ctl_socket = socket.socket(socket.AF_BLUETOOTH, socket.SOCK_SEQPACKET,
AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'
```

furlongm, 2021-04-27:
On OpenSUSE, it seems that python is compiled without bluetooth support because the compilation introduces a circular dependency.

https://lists.opensuse.org/archives/list/factory@lists.opensuse.org/message/J65TRIE7UA5G37DWEGRRW7GH2KISLY62/

The solution suggested in that thread is to use pybluez BluetoothSocket:

https://pybluez.readthedocs.io/en/latest/api/bluetooth_socket.html