chrippa/ds4drv #151: ds4drv command not found

mouarfff, 2018-06-29:
I tried to install ds4drv on linux mint mate 18.3 but i have problems

```
pegasus@andromeda ~ $ sudo -H pip3 install ds4drv
Requirement already satisfied: ds4drv in /usr/local/lib/python3.5/dist-packages (0.5.1)
Requirement already satisfied: evdev>=0.3.0 in /usr/local/lib/python3.5/dist-packages (from ds4drv) (1.0.0)
Requirement already satisfied: pyudev>=0.16 in /usr/lib/python3/dist-packages (from ds4drv) (0.16.1)
```

But there is no ds4drv binary

Alkaris, 2018-09-13:
You're doing it all wrong just do this.;
`sudo pip install ds4drv`

it should install and you'll be able to run it, just type `ds4drv -h` into terminal.

Also it looks like it's already installed by what you're giving in your output there, just do `ds4drv --version` into Terminal and it'll show it's version, it should report back `0.5.1` if it's installed.

mouarfff, 2018-09-20:
I tried this to, but it didn't work ...

I funded a better solution ..... :(
 stopping linux after 13 years using it :(

cbaez440, 2019-08-26:
`sudo pip3 install ds4drv` worked for me

closed 2019-11-21 by Ape