chrippa/ds4drv #160: Install requires gcc

CameronDevine, 2018-10-30:
Today I was attempting to install `ds4drv` on a new installation of ubuntu mate 18.04. I had a strange error message showing up, which I was eventually able to fix by installing `gcc` on my system. Should this be added to the dependencies listed in the readme?

catscarlet, 2019-01-10:
Did you already install python-dev before gcc? And where is your 'strange error message'?

CameronDevine, 2019-01-10:
I was able to reproduce this error, and I am seeing a very clear error message, so I don't know why I thought this was so confusing.

```
unable to execute 'x86_64-linux-gnu-gcc': No such file or directory
error: Setup script exited with error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
```

What I have found is that if you install `pip` through the Ubuntu repositories it installs `gcc` automatically. However, to get the most recent version of `pip` I had downloaded it from `https://bootstrap.pypa.io/get-pip.py`, which of course does not automatically install `gcc`.

catscarlet, 2019-01-10:
How curious. How could a 100% python project need gcc.

CameronDevine, 2019-01-10:
There are a few C files that need to be compiled for [`python-evdev`](https://github.com/gvalkov/python-evdev/tree/master/evdev).

catscarlet, 2019-01-10:
OK I just got another bug and I checked the issue list and find it already got fixed and merged, but saddly the last released is 0.5.1 in 2016.
Seems not be properly maintained properly. Lots PRs there.

Send PR if you think gcc is required in README.MD

CameronDevine, 2019-01-10:
I don't think it is necessary. Most people will have `gcc` installed when they install `pip`, and the error message is reasonably clear.

catscarlet, 2019-01-11:
If not, what's the point of this issue?

CameronDevine, 2019-01-14:
This issue was opened because I thought there was a point three months ago. Based on what I know now, I don't think I would ever have opened an issue.