chrippa/ds4drv #163: [missing] Model CUH-ZCT2 [DualShock 4 v2]

Cevastes, 2019-03-27:
Hi,

It seems the newest revision to the DualShock 4, CUH-ZCT2, is unrecognized by ds4drv

Su20ma, 2019-06-28:
ds4drv development was is halted. there haven't been any changes for 3 years supporting new DS4 might not happen

cslev, 2020-09-15:
I have CUH-ZCT2G (2nd gen) controller, and ds4drv recognizes it! I have other issues related to mode changing (issue 181)[https://github.com/chrippa/ds4drv/issues/181], but it seems it is ds4drv/python issue, not the controller's problem

barolo, 2020-10-01:
@cslev I can't make it work with most of the sttuff that isusing sdl2, citra, yuzu, pegasus-frontend. It works natively just not with ds4drv, tested some up to date forks, no go. How is it for you?

cslev, 2020-10-01:
What is your problem now? I actually managed to make my controller fully work with python 2.7 and the hacks I mentioned in the [issue above](https://github.com/chrippa/ds4drv/issues/181).
I solely use ds4drv and installed only the required binaries/libraries and don't use any system-level bluetooth manager.

Actually I built a small gaming PC with PopOS, and ds4drv is launched at startup from crontab and it works like a charm with all modes.
The only problem is that with Proton it not really works, but with Wine it is working (dunno whether it is a ds4drv issue, though).

barolo, 2020-10-01:
@cslev my problem is that it works with some stuff and with some doesn't. I'm trying to do a similar thing; a gaming pc with a bunch of games, emus behind a one frontend. The thing is, the frontend doesn't work, and some emus too.
Could you perhaps test if this works for you? Just browsing the UI with a pad it's a small download https://github.com/mmatyas/pegasus-frontend/releases/tag/alpha14

cslev, 2020-10-02:
My controller never rumbled :D 
The LED colors are anyway f***ed up at some point at my side, since if you have multiple controllers, how can you define different colors schemes for different modes (kbmouse,default,xpad) for each controller??
BTW., which game do you use that supports rumble? I want to test it and see how it works! thanks

barolo, 2020-10-02:
I think that it depends on which fork you're using, it works with everything that has rumble either natively or via bluetooth with  [ds4drv-cemuhook](https://github.com/TheDrHax/ds4drv-cemuhook) in hidraw mode, currentlly I'm playing Ori and some Wii U games via Cemu, Wii games via dolphin [ that one would be easy to test since it has tester in the options ] wine/proton stuff works too @cslev

cslev, 2020-10-02:
ohhh, very new information received! So, you did not use the ds4drv via pip install and not even the chrippa version.
What is this cemuhook version! This is actually the same but having rumble support? or is it a extension to this chrippa version?
Sorry, so many forks with the same README.md files, I am getting confused... thanks

barolo, 2020-10-02:
It's the same but with added functionality and maintained, cemuhook part  is for motion input via --udp flag, only few emulators use that, like cemu, citra and yuzu.
With wine/proton/lutris it works out of the box for me
Also, that [script in the comment](https://gaming.stackexchange.com/questions/336934/how-to-set-default-color-and-brightness-of-leds-of-the-dualshock-4-controller-on) works for setting the colour. I mainly need it for lowering the brightness

cslev, 2020-10-02:
ohh, okay! Then, I try this version with Wine! thanks

barolo, 2020-10-02:
also you can use pip install [ straight from git ] with that fork, think it's in the readme too

cslev, 2020-10-02:
and which python version do you suggest? python 3.8 is okay or too new? python2.7 also works?

barolo, 2020-10-02:
I don't even have python2 on my system, I'm using it with 3.8

cslev, 2020-10-02:
> It's the same but with added functionality and maintained, cemuhook part is for motion input via --udp flag, only few emulators use that, like cemu, citra and yuzu.
> With wine/proton/lutris it works out of the box for non emu stuff.
> Also, that [script in the comment](https://gaming.stackexchange.com/questions/336934/how-to-set-default-color-and-brightness-of-leds-of-the-dualshock-4-controller-on) works for setting the colour. I mainly need it for lowering the brightness
> I have no need really for other profiles since touchapd works and I'm using that as mouse if needed.

I checked this script, I was not talking about this issue. ds4drv has also the built-in LED color change for different controllers. But when you change modes (from default to xpad, for instance) then there are built-in LED colors in the config file! I don't know/did not find any info, how LED coloros for the different modes for different controllers can be set! But, it is not a big issue, I can live without it....I anyway use one controller most of the time.

barolo, 2020-10-02:
The problem is that with BT/hidraw leds don't work, there are some patches/pull requests but at least with some apps it gets disabled with rumble [ I have a suspicion as to why, probably doesn't happen with normal games ].
And as said I have no need for the profiles at all.