chrippa/ds4drv #164: Does not support model CUH-ZCT2E

kylebarbour, 2019-03-31:
**Issue:** `ds4drv` does not recognize some of the newer DS4 devices, including CUH-ZCT2E.

**Expected behavior:** `ds4drv` would recognize this controller like any other.

**Potentially related bugs:** #104, #114, #162, #163

**Related info:** My controller shows up using `lsusb` as `054c:05c4 Sony Corp. DualShock 4`. My output when using `ds4drv`:
    [info][controller 1] Created devices /dev/input/js1 (joystick) /dev/input/event16 (evdev) 
    [info][bluetooth] Scanning for devices
At this point it hangs. Same thing when using `hidraw` via USB:
    $ ds4drv --hidraw
    [info][controller 1] Created devices /dev/input/js1 (joystick) /dev/input/event16 (evdev) 
    [info][hidraw] Scanning for devices

**My system:**
    $ lsb_release -cdr:
    Description:	Ubuntu 18.04.2 LTS
    Release:	18.04
    Codename:	bionic
   $ uname -rs
    Linux 4.15.0-46-generic

kylebarbour, 2019-03-31:
Looks like this was from accidentally buying a knockoff controller. Using one that I'm more confident was an official controller works fine. Thanks!

closed 2019-03-31 by kylebarbour

AryToNeX, 2020-03-11:
the CUH-ZCT2E is not a knockoff controller, and I can confirm that it does not work.
How do I know that? Well, my PS4 came with it. Also, if you search the Sony website, there's even an user manual for that model.

AradiaIsStupid, 2022-03-28:
is there any updates about this? i am experiencing the exact same issue

noabody, 2022-04-23:
On Arch, I modified the [hid-sony-ds3usb-dkms](https://aur.archlinux.org/packages/hid-sony-ds3usb-dkms#comment-861015) package with the [patch](https://lkml.org/lkml/2021/1/13/853) that chloekek referenced so that my Wingman XE adapter could be recognized as a DS4 controller.  This adapter, and I presume many knock-off, or alternative controllers, use feature report 0x12 instead of 0x81.  I think adding support to 0x12 in ds4drv would be of great benefit to users.

See more information at my [blog](https://noabody.wordpress.com/2022/04/15/brook-wingman-xb-xe/).

noabody, 2022-05-14:
I picked up a [knock-off](https://www.aliexpress.com/item/1005003957459405.html) ds4 controller and it identifies itself as 054C:09CC which is a CUH-ZCT2x variant.  It's device name would match ds4drv information, same as noted above.

Without the custom software from my previous message, here is the dmesg:
```
usb 8-1.2: new full-speed USB device number 11 using xhci_hcd
usb 8-1.2: New USB device found, idVendor=054c, idProduct=09cc, bcdDevice= 1.00
usb 8-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 8-1.2: Product: Wireless Controller
usb 8-1.2: Manufacturer: Sony Interactive Entertainment
sony 0003:054C:09CC.000C: failed to retrieve feature report 0x81 with the DualShock 4 MAC address
sony 0003:054C:09CC.000C: hidraw9: USB HID v81.11 Gamepad [Sony Interactive Entertainment Wireless Controller] on usb-0000:00:10.1-1.2/input3
sony 0003:054C:09CC.000C: failed to claim input
```
And here it is with my hid-sony-ds3usb-dkms package that I build for Arch Linux.

```
usb 8-1.2: new full-speed USB device number 12 using xhci_hcd
usb 8-1.2: New USB device found, idVendor=054c, idProduct=09cc, bcdDevice= 1.00
usb 8-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 8-1.2: Product: Wireless Controller
usb 8-1.2: Manufacturer: Sony Interactive Entertainment
input: Sony Interactive Entertainment Wireless Controller Touchpad as /devices/pci0000:00/0000:00:10.1/usb8/8-1/8-1.2/8-1.2:1.3/0003:054C:09CC.000E/input/input47
input: Sony Interactive Entertainment Wireless Controller Motion Sensors as /devices/pci0000:00/0000:00:10.1/usb8/8-1/8-1.2/8-1.2:1.3/0003:054C:09CC.000E/input/input48
input: Sony Interactive Entertainment Wireless Controller as /devices/pci0000:00/0000:00:10.1/usb8/8-1/8-1.2/8-1.2:1.3/0003:054C:09CC.000E/input/input46
sony 0003:054C:09CC.000E: input,hidraw9: USB HID v81.11 Gamepad [Sony Interactive Entertainment Wireless Controller] on usb-0000:00:10.1-1.2/input3
```
So that patch works for any generic controller, especially those that identify themselves with a proper VID:PID.  I can work for other similarly capable devices as long as their VID:PID is patched in as was noted in my blog.

AradiaIsStupid, 2022-05-18:
i am sorry but can you please upload the patched files? thank you!

noabody, 2022-05-18:
I can't because the method I used turns a kernel module into dkms.  You can extract my arch [patch](https://raw.githubusercontent.com/noabody/unibuild/master/data/arch/hid-sony-ds3usb-dkms.patch) with commands:
```
patch -Np1 -i hid-sony-ds3usb-dkms.patch
patch -Np1 -i hid-sony-ds3usb.patch
```
You'd still need appropriate kernel sources, the latest being:
```
wget https://github.com/torvalds/linux/raw/master/drivers/hid/hid-sony.c
wget https://github.com/torvalds/linux/raw/master/drivers/hid/hid-ids.h
```
Ubuntu has some pointers on how to build a [dkms](https://help.ubuntu.com/community/DKMS) package.

It's going to require some reverse engineering no matter how you look at it.  That's what I had to do.

AradiaIsStupid, 2022-05-18:
thank you it worked :D

BlessedBoi, 2022-12-04:
> I can't because the method I used turns a kernel module into dkms. You can extract my arch [patch](https://raw.githubusercontent.com/noabody/unibuild/master/data/arch/hid-sony-ds3usb-dkms.patch) with commands:
> 
> ```
> wget https://github.com/noabody/unibuild/raw/master/data/arch/hid-sony-ds3usb-dkms.patch
> patch -Np1 -i hid-sony-ds3usb-dkms.patch
> patch -Np1 -i hid-sony-ds3usb-dkms/hid-sony-ds3usb-dkms.patch
> ```
> 
> You'd still need appropriate kernel sources, the latest being:
> 
> ```
> wget https://github.com/torvalds/linux/raw/master/drivers/hid/hid-sony.c
> wget https://github.com/torvalds/linux/raw/master/drivers/hid/hid-ids.h
> ```
> 
> Ubuntu has some pointers on how to build a [dkms](https://help.ubuntu.com/community/DKMS) package.
> 
> It's going to require some reverse engineering no matter how you look at it. That's what I had to do.

Thanks, works great. But is there opportunity to build something like this for bluetooth connection?

chloekek, 2022-12-04:
CUH-ZCT2E already works fine with Bluetooth. The problem is USB-specific.

BlessedBoi, 2022-12-04:
> CUH-ZCT2E already works fine with Bluetooth. The problem is USB-specific.

So you can for example change color of light bar using steam?

davidalxdre, 2023-02-22:
> I can't because the method I used turns a kernel module into dkms. You can extract my arch [patch](https://raw.githubusercontent.com/noabody/unibuild/master/data/arch/hid-sony-ds3usb-dkms.patch) [...]

Hey bro, could you help me, please?

i have a steam deck and i came to this github because of this thread on reddit:

https://www.reddit.com/r/SteamDeck/comments/zcfun1/comment/j8ivf5s/

My dualshock has the same problem as mentioned,and i tried to install these packages, i had to use pacman to install in the SteamOS, but the controllers still not working after paired, there is a another way to install these? or there is a way to find out whats it's going on after paired?

thank you in advanced.

AradiaIsStupid, 2023-02-22:
> Hey bro, could you help me, please?
>
> i have a steam deck and i came to this github because of this thread on reddit:
>
> https://www.reddit.com/r/SteamDeck/comments/zcfun1/comment/j8ivf5s/
>
> My [...]
can you connect it with a cable and type `dmesg` in the terminal and show what it says after it's connected?