chrippa/ds4drv #166: third party controllers

anonronin, 2019-05-28:
i bought a third party controller but it just "[info][hidraw] Scanning for devices"
I did the udev rules and lsmod | grep uinput nothing but it seems it's built-in
cat /boot/config-5.0.0-rc4-exton | grep UINPUT
CONFIG_INPUT_UINPUT=y
i have /dev/uinput
Bus 004 Device 005: ID 054c:05c4 Sony Corp. DualShock 4 [CUH-ZCT1x]
it should work or not? 
thanks in advance

Su20ma, 2019-06-28:
What distro are you using? it should work if you enable hidraw and it looks you did I suggest using bluetooth if available

anonronin, 2020-10-18:
An original Sony ds4 did the trick, thank you so much regards

albfan, 2021-12-16:
I use:

    Bus 001 Device 007: ID 054c:05c4 Sony Corp. DualShock 4 [CUH-ZCT1x]

and works correctly on linux. I think this issue can be closed