chrippa/ds4drv #167: gcc: fatal error: cannot read spec file ‘/usr/lib/rpm/redhat/redhat-hardened-cc1

orwell1984, 2019-08-16:
pip install fails with `gcc: fatal error: cannot read spec file ‘/usr/lib/rpm/redhat/redhat-hardened-cc1’: No such file or directory`

I'm running Fedora 30.

A little side question - how does this driver "cooperate" with Fedora's native PS4 support and Steam's settings for controllers? Are they mutually exclusive?

CoryParsnipson, 2019-09-30:
I got the same error (I'm also running Fedora 30). I found this on stack overflow:

https://stackoverflow.com/questions/34624428/g-error-usr-lib-rpm-redhat-redhat-hardened-cc1-no-such-file-or-directory

That says this error can be resolved by installing redhat-rpm-config. (sudo dnf install redhat-rpm-config).

After that, your install might fail with "evdev/input.c:10:10: fatal error: Python.h: No such file or directory". If you install the python2-devel package, the build should pass after that. (sudo dnf install python-devel)

Scotchman0, 2021-05-25:
hello from the future! Just chipping in to say thank you for the python-devel tip - was what I was missing. (Fedora 34). appreciate it!

JPDucky, 2022-10-13:
I ran into this trying to install revdev via `sudo gem install revdev` and ended up with the error:
`[palmerd@fedora ~]$ sudo gem install revdev
Fetching revdev-0.2.1.gem
Building native extensions. This could take a while...
ERROR:  Error installing revdev:
        ERROR: Failed to build gem native extension.

    current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
/usr/bin/ruby -I /usr/share/rubygems -r ./siteconf20221012-57856-bh1fg.rb extconf.rb
mkmf.rb can't find header files for ruby at /usr/share/include/ruby.h

You might have to install separate package for the ruby development
environment, ruby-dev or ruby-devel for example.

extconf failed, exit code 1

Gem files will remain installed in /usr/local/share/gems/gems/revdev-0.2.1 for inspection.
Results logged to /usr/local/lib64/gems/ruby/revdev-0.2.1/gem_make.out`

-Okay, let me install ruby devel with `sudo dnf install ruby-devel`, then again:
`[palmerd@fedora ~]$ sudo gem install revdev
Building native extensions. This could take a while...
ERROR:  Error installing revdev:
        ERROR: Failed to build gem native extension.

    current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
/usr/bin/ruby -I /usr/share/rubygems -r ./siteconf20221012-57926-awmgln.rb extconf.rb
creating Makefile

current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
make DESTDIR\= clean
rm -f 
rm -fr revdev.so false *.o  *.bak mkmf.log .*.time

current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
make DESTDIR\=
gcc -I. -I/usr/include -I/usr/include/ruby/backward -I/usr/include -I. -DRUBY_1_9    -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -m64 -o revdev.o -c revdev.c
gcc: fatal error: cannot read spec file ‘/usr/lib/rpm/redhat/redhat-hardened-cc1’: No such file or directory
compilation terminated.
make: *** [Makefile:247: revdev.o] Error 1

make failed, exit code 2

Gem files will remain installed in /usr/local/share/gems/gems/revdev-0.2.1 for inspection.
Results logged to /usr/local/lib64/gems/ruby/revdev-0.2.1/gem_make.out`

and the contents of /usr/local/lib64/gems/ruby/revdev-0.2.1/gem_make.out:
`[palmerd@fedora revdev-0.2.1]$ cat gem_make.out 
current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
/usr/bin/ruby -I /usr/share/rubygems -r ./siteconf20221012-57926-awmgln.rb extconf.rb
creating Makefile

current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
make DESTDIR\= clean
rm -f 
rm -fr revdev.so false *.o  *.bak mkmf.log .*.time

current directory: /usr/local/share/gems/gems/revdev-0.2.1/ext/revdev
make DESTDIR\=
gcc -I. -I/usr/include -I/usr/include/ruby/backward -I/usr/include -I. -DRUBY_1_9    -fPIC -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fPIC -m64 -o revdev.o -c revdev.c
gcc: fatal error: cannot read spec file ‘/usr/lib/rpm/redhat/redhat-hardened-cc1’: No such file or directory
compilation terminated.
make: *** [Makefile:247: revdev.o] Error 1

make failed, exit code 2`

then I found this thread. I installed redhat-rpm-config then ran 'sudo gem install revdev` and it installed without errors.

I had previously installed ruby, ydotool, libevdev-devel, @development-tools, and ruby-devel prior to getting this to work. I hope this helps someone.