chrippa/ds4drv #170: Can't find DS4 device(ps4 pro) with USB cable/--hidraw

trialism, 2019-10-20:
```
[pc trialism]# ds4drv --hidraw
[info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event23 (evdev)
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices
```

I'm on a fresh manjaro linux. I tried with bluetooth too but I got this:

`[error][daemon] 'hcitool' could not be found, make sure you have bluez-utils installed.`

bluez-utils is installed.

gfacchini, 2020-04-10:
Same issue. Fresh install under Ubuntu with KDE.

goremukin, 2020-07-05:
Same issue. Linux Mint 19

goremukin, 2020-07-05:
Related to this issue:
https://github.com/chrippa/ds4drv/issues/104

goremukin, 2020-07-05:
I found solution. You need to edit `/usr/local/lib/python2.7/dist-packages/ds4drv/backends/hidraw.py` like here:
https://github.com/chrippa/ds4drv/pull/105/files

bss233, 2021-09-16:
> /usr/local/lib/python2.7/dist-packages/ds4drv/backends/hidraw.py

If you have python3.* installed, you may have installed ds4drv with pip3, so change the path slightly to `/usr/local/lib/python3.*/dist-packages/ds4drv/backends/hidraw.py`

where * is whatever version of Python 3 you're on

albfan, 2021-12-16:
hcitool is an external tool you need to install.

Probably use bluetoothctl would make sense these days.

My controller shows: Sony Computer Entertainment Wireless controller, so maybe check case insensitive could help. Still I cannot connect by usb