chrippa/ds4drv #172: Auto Connect / auto start

jdl159, 2019-12-24:
So I’ve followed all directions to include adding “ /usr/local/bin/ds4drv & “ to the rc.local file by going to “ sudo nano /etc/rc.local” and saved it with ctr+o /etc/rc.local 

BUT whenever I reboot the Raspberry Pi the drivers / ds4drv won’t start unless I manually start it from the terminal with a “Sudo Ds4drv”

I’ve checked by running  “jstest /dev/input/js0” in terminal when I reboot the Pi and attempt to pair the controller. 

And 

I’ve verified the /etc/rc.local file retained the changes and still has the “/usr/local/bin/ds4drv & “ line in it. 

Only after doing “Sudo ds4drv” will the controller connect and then jstest work. 
Not to sure why this is happening. I’ve tried to trouble shoot it to the best of my own abilities.

I’d like to get it to automatically connect if a controller is in pairing mode.

s-aguado, 2020-01-26:
Same issue here, I solved it creating a systemd service. 

1. Create unit file (`sudo nano /lib/systemd/system/ds4drv.service`) with the following content
```
[Unit]
Description=Start ds4drv after boot. Daemon to pair PS4 bluetooth controllers. 
After=multi-user.target
Requires=bluetooth.target

[Service]
Type=idle
ExecStart=/usr/local/bin/ds4drv --led=000008
Restart=always

[Install]
WantedBy=multi-user.target
```
2. Load the changes and enable the service
```
sudo systemctl daemon-reload
sudo systemctl enable ds4drv.service
```
3. Try now
```
sudo reboot
```

(I've written it down from memory, forgive the possible misprints.)

andrewtquick, 2020-09-14:
> Same issue here, I solved it creating a systemd service.
> 
> 1. Create unit file (`sudo nano /lib/systemd/system/ds4drv.service`) with the following content
> 
> ```
> [Unit]
> Description=Start ds4drv after boot. Daemon to pair PS4 bluetooth controllers. 
> After=multi-user.target
> Requires=bluetooth.target
> 
> [Service]
> Type=idle
> ExecStart=/usr/local/bin/ds4drv --led=000008
> Restart=always
> 
> [Install]
> WantedBy=multi-user.target
> ```
> 
> 1. Load the changes and enable the service
> 
> ```
> sudo systemctl daemon-reload
> sudo systemctl enable ds4drv.service
> ```
> 
> 1. Try now
> 
> ```
> sudo reboot
> ```
> 
> (I've written it down from memory, forgive the possible misprints.)

+1 Definitely solved my problem. I made one slight change:

`ExecStart=sudo /usr/local/bin/ds4drv --led=000008`

ukos-git, 2021-04-02:
No need to do a reboot after enabling the service. 

``` 
sudo systemctl start ds4drv.service
sudo systemctl status ds4drv.service
```

I would prefer the above configuration with the an additional  ``--hidraw``  flag and then the file from this issue should go to the repo [systemd/ds4drv.service](https://github.com/chrippa/ds4drv/blob/master/systemd/ds4drv.service) with a PR. The (official) service proposal there wasn't touched the last 7y.

I am unsure but user space systemd services should go to `/usr/local/lib/systemd/system` so they do not break on system upgrades. Although, I think the naming is fairly unique and it can probably stay the way it is ` /lib/systemd/system/ds4drv.service`.

My proposal for the file:

```
# Source: https://github.com/chrippa/ds4drv/blob/master/systemd/ds4drv.service
# Location: /lib/systemd/system/ds4drv.service
# Installation: 
# - sudo systemctl daemon-reload
# - sudo systemctl enable ds4drv.service
# - sudo systemctl start ds4drv.service
# - sudo systemctl status ds4drv.service

[Unit]
Description=Pair DS4 bluetooth controllers.
Requires=bluetooth.service
After=multi-user.target

[Service]
ExecStart=/usr/local/bin/ds4drv --hidraw
Restart=on-abort

[Install]
WantedBy=multi-user.target
```

albfan, 2021-12-16:
I think this service is included on repo, so it can be closed