chrippa/ds4drv #173: Cannot connect to DS4 with Bluetooth on Fedora

bigtobster, 2019-12-31:
Hi

**OS:** Fedora 30
**Kernel:** 5.3.16-200.fc30.x86_64
**BT:** HCI Version: 4.0 (0x6)  Revision: 0x500
	LMP Version: 4.0 (0x6)  Subversion: 0x500
**BlueZ:** bluez-5.52-1

DS4 connects and works perfectly through USB with all features. Tested via Parsec.

DS4 connects fine through Blueberry Tray using the GUI. Issues an address and everything. Parsec does **not** receive any inputs although it does recognise that a DS4 has been connected. 

**DS4DRV Log:**
`[info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event15 (evdev) 
[info][hidraw] Scanning for devices
[info][hidraw] Scanning for devices`

**DMESG Log**
```
[ 4686.659570] sony 0005:054C:05C4.0006: unknown main item tag 0x0
[ 4686.660055] sony 0005:054C:05C4.0006: DualShock 4 calibration report's CRC check failed, received crc 0x1cffff91 != 0xd811a775
[ 4686.660060] sony 0005:054C:05C4.0006: Retrying DualShock 4 get calibration report request
[ 4686.660068] sony 0005:054C:05C4.0006: DualShock 4 calibration report's CRC check failed, received crc 0x1cffff91 != 0xd811a775
[ 4686.660072] sony 0005:054C:05C4.0006: Retrying DualShock 4 get calibration report request
[ 4686.660079] sony 0005:054C:05C4.0006: DualShock 4 calibration report's CRC check failed, received crc 0x1cffff91 != 0xd811a775
[ 4686.660082] sony 0005:054C:05C4.0006: Failed to get calibration data from Dualshock 4
[ 4686.660403] sony 0005:054C:05C4.0006: hidraw0: BLUETOOTH HID v81.00 Gamepad [Wireless Controller] on fc:f8:ae:69:b6:02
[ 4686.660409] sony 0005:054C:05C4.0006: failed to claim input
```

Dmesg indicates possible issue on device resolution/verification?

arael, 2020-02-17:
I have the same issue.

OS: Ubuntu 19.10
Kernel: 5.3.0-29-generic
BlueZ: 5.50-0ubuntu5

ror6ax, 2020-03-15:
@bigtobster does rumble work for you over usb?

bigtobster, 2020-03-15:
> @bigtobster does rumble work for you over usb?

Yes works fine over USB with Steam

albfan, 2021-12-16:
I play a bit with dkms and kernel before use ds4. That error do not prevent ds4 to work correctly.

Anyway is annoying, and If I can get rid of it will try to report here.