chrippa/ds4drv #182: DualSense Support?

cow-killer, 2020-11-17:
Hey there, I don't suppose there's any way we could reverse-engineer the DualSense controller's connectivity and get this working within `ds4drv`?

AurelienKun, 2021-04-06:
I second this demand. Ive tried a kernel 5.12 that enable the driver but somehow the mapping is still wrong and the gamepad does not work at all on some emulators like dolphin. 

The driver implemented by steam (whatever the kernel version) seems to work very well though (but only for steam games).
It might be great to look at their implementation.

QuixoticGitHub, 2021-08-30:
Status?

albfan, 2021-12-16:
If https://github.com/naoki-mizuno/ds4_driver/issues/16 was closed, what's the point of this issue?

this is ds4 (dualshock), why would it support dualsense (ds5)?

barkovv, 2022-01-05:
Dualsense (ds5) is supported out of box by hid-playstation module from linux kernel (starting from version 5.12, [here](https://patchwork.kernel.org/project/linux-input/patch/20201219062336.72568-2-roderick@gaikai.com/) are the patches for it ).

Both bluetooth and USB mode are supported by Sony. I don't see any reason for opensource community driver.

QuixoticGitHub, 2022-01-05:
Hello people. Regarding 5.12 support for DualSense it seems that the driver
released by Sony was either containing a few errors or something went wrong
during the implementation. From what I recall it was the vertical axis of
the left stick button and the R2 button that was swapped.

The controller works through Steam... But only because they use their own
driver.

Any other native Linux game (tuxracer etc) only works if users manage to
configure the controllers within the games and that hardly should be
necessary.

AurelienKun, 2022-01-05:
@barkovv try something outside Steam plateform and you'll see the issue (whatever you're using BT or USB).

QuixoticGitHub, 2022-01-05:
Simplest demonstration... Install jstest-gtk and see for yourselves :)

barkovv, 2022-01-06:
Yeah, you're right now I see the problem in the NonSteam environment (in Cyberpunk 2077 via Geforce Now).

I found a workaround for it using [xboxdrv](https://xboxdrv.gitlab.io/) (from [here](https://github.com/yoyossef/ds360))
Consider that you have [dualsense.xboxdrv](https://github.com/yoyossef/ds360/blob/main/dualsense.xboxdrv)  (this is mine example, I've just modified evdev value):
```
[xboxdrv]

#evdev = /dev/input/by-id/usb-Sony_Interactive_Entertainment_Wireless_Controller-if03-event-joystick
evdev = /dev/input/event15
evdev-grab = true
evdev-debug = true
mimic-xpad = true
silent = true
quiet = true
#deadzone = 5%
#deadzone-trigger = 10%

[axismap]

-Y1 = Y1
-Y2 = Y2

[evdev-absmap]

ABS_HAT0X = dpad_x
ABS_HAT0Y = dpad_y
ABS_X     = X1
ABS_Y     = Y1
ABS_RX    = X2
ABS_RY    = Y2
ABS_Z     = LT
ABS_RZ   = RT

[evdev-keymap]

BTN_SOUTH    = A
BTN_EAST     = B
BTN_NORTH    = Y
BTN_WEST     = X
BTN_START    = start
BTN_MODE     = guide
BTN_SELECT   = back
BTN_TL       = LB
BTN_TR       = RB
BTN_TL2      = LT
BTN_TR2      = RT
BTN_THUMBL   = TL
BTN_THUMBR   = TR

# EOF #
```

evdev value was found in _/proc/bus/input/devices_ (here is part of mine for Bluetooth connection)
```
I: Bus=0005 Vendor=054c Product=0ce6 Version=8100
N: Name="Wireless Controller"
P: Phys=
S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/bluetooth/hci0/hci0:256/0005:054C:0CE6.0003/input/input24
U: Uniq=d0:bc:c1:98:17:63
H: Handlers=event15 js0
B: PROP=0
B: EV=20000b
B: KEY=7fdb000000000000 0 0 0 0
B: ABS=3003f
B: FF=107030000 0
```
For my case it's _/dev/input/*event15*_ (replace with yours if needed)
Then you should just do:
```bash
sudo xboxdrv -c dualsense.xboxdrv
```

It should fix the problem (works for me). In case it's not just try to delete **/dev/input/*js0* file (js0 is taken from above, replace with yours one if needed).

Hope it helps.

QuixoticGitHub, 2022-01-06:
Question is, to whom do we forward this? Seems the support rolled out in
5.12 needs adjustment.

QuixoticGitHub, 2022-02-03:
Did anyone else manage to confirm/fix this? 

Do we push this information to someone else?

loveisfoss, 2022-02-18:
> If [naoki-mizuno/ds4_driver#16](https://github.com/naoki-mizuno/ds4_driver/issues/16) was closed, what's the point of this issue?
> 
> this is ds4 (dualshock), why would it support dualsense (ds5)?

The point would be to create a dsu server in order to use motion controls with cemu (which I personally run with wine/lutris).