chrippa/ds4drv #183: Can't change LED colour

Foro38, 2020-11-24:
When I type `ds4drv --led ff0000` in the terminal i get this:

Traceback (most recent call last):
  File "/usr/bin/ds4drv", line 33, in <module>
    sys.exit(load_entry_point('ds4drv==0.5.1', 'console_scripts', 'ds4drv')())
  File "/usr/lib/python3.8/site-packages/ds4drv/__main__.py", line 132, in main
    options = load_options()
  File "/usr/lib/python3.8/site-packages/ds4drv/config.py", line 199, in load_options
    config.load(path)
  File "/usr/lib/python3.8/site-packages/ds4drv/config.py", line 74, in load
    self.read([filename])
  File "/usr/lib/python3.8/configparser.py", line 697, in read
    self._read(fp, filename)
  File "/usr/lib/python3.8/configparser.py", line 1093, in _read
    raise DuplicateOptionError(sectname, optname,
configparser.DuplicateOptionError: While reading from '/etc/ds4drv.conf' [line 118]: option 'rel_wheelup' in section 'mapping:keyboard' already exists

and the led colour doesn't change

I am using a wired controller on arch linux

albfan, 2021-12-16:
would be interesting to see that `ds4drv.conf` file contents

POOnami, 2023-05-24:
place a "#" in beginning of line 118 
install "hid-sony-panic-fix-dkms"
common problem with knockoff controllers