chrippa/ds4drv #185: Can't connect ps4 controller with ds4drv

poopman97, 2021-05-25:
I type in ds4drv then:
[error] [controller1] Failed to create "/dev/uinput" cannot be opened for writing
[info] [bluetooth] Scanning for devices
[info] [bluetooth] Found device 98:B6:E9:4F:70:17

Then the program crashes
Ibhave linux mint 20.1 kernel version is 5.8

albfan, 2021-12-16:
I think you forget to install udev rules

BooniChan, 2026-10-06:
Same issue here. 
Weirdly enough, if I enable the kernel driver (hid_playstation) then ds4drv will recocnize the controller. But I don't think that's the correct way to do it - because if the kernel driver has already taken over I don't think the userspace driver can override it. 
As soon as hid_playstation kernel driver is disabled, ds4drv gets stuck scanning for devices.

Jopp-gh, 2026-10-06:
@BooniChan 
ds4drv is obsolete. 

Also check out if your controller isn't some cheap, unknown brand. Get a Nacon controller on amazon, in case. 

Uninstall any controller drivers you installed yourself and get AntiMicroX to remap any good controller.

BooniChan, 2026-10-06:
@Jopp-gh 
> [@BooniChan](https://github.com/BooniChan) ds4drv is obsolete.
> 
> Also check out if your controller isn't some cheap, unknown brand. Get a Nacon controller on amazon, in case.
> 
> Uninstall any controller drivers you installed yourself and get AntiMicroX to remap any good controller.

My controller is an original playstation 4 controller. It works perfectly fine with bluetooth on games, but there is one game in particular that won't read it. 
The only reason I'm trying to use ds4drv is to emulate a 360 as the game works good with my wired 360 and unfortunately I don't think it can be done with the in kernel hid_playstation driver. 

For other games the controller works out of the box.

Ryochan7, 2026-10-07:
You could probably use xboxdrv to create a 360 controller mapping for the DS4. Here is a link to a supposedly pre-made config.

https://www.reddit.com/r/linux_gaming/comments/10z0dgh/xboxdrv_dualshock_4_preset/

Google also suggests a different mapper program that might work but I never heard of it before.

https://github.com/niku200/ds4to360xboxdrv

If nothing else, you could likely just fall back to using Steam Input.

BooniChan, 2026-10-07:
@Jopp-gh 
I actually tried AntimicroX and it worked.  It does a work around by mapping the controller's keys to the keyboard keys so the game still thinks the keyboard is being used whilst using the ps4 controller.

Thank you for that!

BooniChan, 2026-10-07:
> You could probably use xboxdrv to create a 360 controller mapping for the DS4. Here is a link to a supposedly pre-made config. [...]

@Ryochan7  I'll give it a try. I thought the xboxdrv is a userspace driver for the 360 controller only so it didn't pass my mind to use it on the ds4, that's why I went with ds4drv. 

I can't use the steam input cause there is a weird situation. The game (mortal-shell) was first installed trough lutris. I tried adding it to steam as an external game and transferring the save file, and the controller actually did work, but for some weird reason the DLC gets disabled when ran trough proton instead of lutris. So it solves one issue and causes another. 
I even tried using GE-Proton trough steam but still same issue, and even tried setting lutris to use steam's proton to run it and same thing happens. 
I can't find anything online why the DLC doesn't get loaded on steam when save is transferred so it must be some random rare bug or something. 

What's even weirder is the game has the option to change controller icons from Xbox to Playstation icons, yet it doesn't support the ds4 controller natively, I don't understan why they even put the controller icons in there if they didn't plan to support it lol

BooniChan, 2026-10-07:
@Ryochan7 
I tested xboxdrv with the buttons set as that reddit post you shared and it works. Thank you! 

**In future if anyone finds this post:**
Both antimicrox and xboxdrv can work for games that don't support ds4 out of the box like mortal shell. 
Antimicrox can either use SDL mapping or set the controller to send keyboard keys
Xboxdrv with a config file directly creates a new /dev/input device which the game can use 

Personally I prefer Xboxdrv remapping as that way the game still sees you as using a controller, whilst with antimicrox keyboard keys the game still sees it as a keyboard thus you will get keyboard icons in the game. But both do the job.