chrippa/ds4drv #190: Keeps telling me to "Make sure your bluetooth device is powered up with 'hciconfig hciX up'"

Voreck, 2022-02-10:
So I ran `sudo hciconfig hciX up` after the error. ANd I espected it to work since I've never had problems after that before. But today when I did that and ran `ds4drv` again I get the same error again
```
❯ ds4drv
[info][controller 1] Created devices /dev/input/js0 (joystick) /dev/input/event260 (evdev)
[info][bluetooth] Scanning for devices
[error][bluetooth] Error while scanning for devices: 'hcitool scan' returned error. Make sure your bluetooth device is powered up with 'hciconfig hciX up'.
```
When I run `hcitool scan` manually I get
```
❯ hcitool scan
Scanning ...
Inquiry failed: Device or resource busy
```
I have no knowledge about bluetooth on linux as ds4drv is literally the only time I use bluetooth

Bscout2011, 2022-03-15:
Running `sudo apt install pi-bluetooth` then `sudo reboot` resolved this issue for me.

git-developer, 2023-02-17:
If you're lucky, re-plugging your bluetooth dongle or rebooting solves this problem. Otherwise, this could be a a duplicate of https://github.com/chrippa/ds4drv/issues/123