chrippa/ds4drv #196: Does CUH-ZCT1x has any support?

steveunixson, 2022-07-01:
So I've bought this controller and I can't get it work over usb
`ds4drv --hidraw`

just infinitely scans for devices

 lsusb
 
 `Bus 001 Device 026: ID 054c:05c4 Sony Corp. DualShock 4 [CUH-ZCT1x]`

udev rules

`KERNEL=="uinput", MODE="0666"
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="054c", ATTRS{idProduct}=="05c4", MODE="0666"
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", KERNELS=="0005:054C:05C4.*", MODE="0666"
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="054c", ATTRS{idProduct}=="09cc", MODE="0666"
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", KERNELS=="0005:054C:09CC.*", MODE="0666"
`

karimrir1, 2022-08-05:
Same thing for me, had to use another controller

AbdulRhmanHS, 2023-07-09:
Isn't there any other software to make this unsupported version work?