chrippa/ds4drv #204: configparser.DuplicateOptionError: While reading from '/etc/ds4drv.conf'

Axis4s, 2023-02-25:
Traceback (most recent call last):
  File "/usr/bin/ds4drv", line 33, in <module>
    sys.exit(load_entry_point('ds4drv==0.5.1', 'console_scripts', 'ds4drv')())
  File "/usr/lib/python3.10/site-packages/ds4drv/__main__.py", line 132, in main
    options = load_options()
  File "/usr/lib/python3.10/site-packages/ds4drv/config.py", line 199, in load_options
    config.load(path)
  File "/usr/lib/python3.10/site-packages/ds4drv/config.py", line 74, in load
    self.read([filename])
  File "/usr/lib/python3.10/configparser.py", line 698, in read
    self._read(fp, filename)
  File "/usr/lib/python3.10/configparser.py", line 1097, in _read
    raise DuplicateOptionError(sectname, optname,
configparser.DuplicateOptionError: While reading from '/etc/ds4drv.conf' [line 118]: option 'rel_wheelup' in section 'mapping:keyboard' already exists

dhasial, 2023-08-26:
Same error here. I have no idea what this option is or why it's in the config file by default :(

softestpoly, 2023-09-10:
Same error here, installed from AUR

Ryochan7, 2023-09-10:
Fixed over 7 years ago. Get someone on the AUR to update the package with a fixed ds4drv.conf file.

https://github.com/chrippa/ds4drv/commit/cf20ed070572427c91411f7006084a0e8548ba93