chrippa/ds4drv #205: OSError: [Errno 5] Input/output error

Zeker2908, 2023-03-06:
ds4drv --hidraw  
[info][controller 1] Created devices /dev/input/js1 (joystick) /dev/input/event25 (evdev) 
Traceback (most recent call last):
  File "/usr/bin/ds4drv", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.10/site-packages/ds4drv/__main__.py", line 154, in main
    for device in backend.devices:
  File "/usr/lib/python3.10/site-packages/ds4drv/backends/hidraw.py", line 173, in devices
    yield cls(name=device_name,
  File "/usr/lib/python3.10/site-packages/ds4drv/backends/hidraw.py", line 34, in __init__
    super(HidrawDS4Device, self).__init__(name, addr, type)
  File "/usr/lib/python3.10/site-packages/ds4drv/device.py", line 84, in __init__
    self.set_operational()
  File "/usr/lib/python3.10/site-packages/ds4drv/backends/hidraw.py", line 89, in set_operational
    self.read_feature_report(0x02, 37)
  File "/usr/lib/python3.10/site-packages/ds4drv/backends/hidraw.py", line 63, in read_feature_report
    return fcntl.ioctl(self.fd, op, bytes(buf))
OSError: [Errno 5] Input/output error

xabylr, 2023-05-23:
I have a chines DS4 clone and I was having the same issue. It seems that the feature 0x02 is not supported by it or something like that.

I don't know if this is required for something, but I have just commented out the function call from the penultimate line from the stack trace ( and added a pass in its place for function be valid in python) and it just works!

```python
class HidrawBluetoothDS4Device(HidrawDS4Device):
    __type__ = "bluetooth"

    report_size = 78
    valid_report_id = 0x11

    def set_operational(self):
        #self.read_feature_report(0x02, 37)
        pass
```

Mordius, 2024-01-28:
Sorry I'm a noob in this, how do I make it work?

xabylr, 2024-01-28:
Hello, if you have **exactly the same error**, look for where the file `hidraw.py` is located in your error trace. Then, open it with a text editor (you may need to use `sudo` to be able to save it) and look for `self.read_feature_report(0x02, 37)`. Comment out this line by putting a hash `#` symbol before the text. Finally, insert a line below with the text `pass`. Save the file, execute the program again and this error should be gone.

What we are doing here is disabling reading a characteristic not implemented in a chinese DS4 remote. The program will try to call the function but it will not do anything.

Keep in mind that this is Python, so in for the above to work, the `pass` word must have more indentation than `def set_operational(self):`