chrippa/ds4drv #207: Gyro mapping

DewDgi, 2024-01-24:
Hi, i used ds4windows, and there i could bind gyro to left stick, as wheel, and set 270 degrees. Like this: **https://www.reddit.com/r/DS4Windows/comments/v6j3ye/help_with_gyro_config_for_racing_games/**. Can i do something like this in ds4drv?

Ryochan7, 2024-01-24:
It looks like ds4drv has not been updated in over 7 years. As for the original question, I am pretty sure the answer is that there is no way to make that type of mapping in ds4drv.

I had not seen any project besides DS4Windows offer that type of mapping option for the Gyro / Accelerometer. That portion of the program was all the work of **mika-n**. I mainly ended up using it to play American Truck Simulator but I played through a couple of racing games using that option in a profile; I do not play racing games much but Dirt Showdown was one racing game that I beat using the Gyro Steering Wheel option.

DewDgi, 2024-01-24:
Do you know any way to do same thing on linux or it's completely pointless?

closed 2024-01-25 by DewDgi