chrippa/ds4drv #208: AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?

BrunoRochaDev, 2024-03-28:
Exception when trying to run. 

```
user@machine:~$ ds4drv
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 5, in <module>
    from ds4drv.__main__ import main
  File "/usr/local/lib/python3.12/site-packages/ds4drv/__main__.py", line 5, in <module>
    from .actions import ActionRegistry
  File "/usr/local/lib/python3.12/site-packages/ds4drv/actions/__init__.py", line 1, in <module>
    from ..action import ActionRegistry
  File "/usr/local/lib/python3.12/site-packages/ds4drv/action.py", line 1, in <module>
    from .config import add_controller_option
  File "/usr/local/lib/python3.12/site-packages/ds4drv/config.py", line 72, in <module>
    class Config(configparser.SafeConfigParser):
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
```

Installed using `sudo pip install ds4drv`.

Fedora Workstation 39. Python Python 3.12.2.

Kompreya, 2024-04-01:
The solution to this is to navigate to `/usr/local/lib/python3.12/site-packages/ds4drv/config.py`and modify lines 72 and 99 from `SafeConfigParser` to `ConfigParser`

Save and run again.

Also, on Fedora, you may run into an error regarding hcitools. If that's the case, then you need to install bluez-deprecated off the rpm, which includes hcitools.

SpiningMonkey, 2025-04-21:
> The solution to this is to navigate to `/usr/local/lib/python3.12/site-packages/ds4drv/config.py`and modify lines 72 and 99 from `SafeConfigParser` to `ConfigParser`
> 
> Save and run again.
> 
> Also, on Fedora, you may run into an error regarding hcitools. If that's the case, then you need to install bluez-deprecated off the rpm, which includes hcitools.
> 
> ds4drv seems to be very old and has not been updated for modern versions of python or bluez, but should still work nonetheless.

Hello, ı have the same ıssue. I want to try this but it's in root file and ı can't access it. İs there a way to access this file? Thanks for help.