chrippa/ds4drv #213: Don't work ds4drv on Ubuntu 22.4

Cebastion, 2025-01-07:
Hello there!

I recently buy Dualshock 4 and I want to it emulate Xbox 360 for Borderlands and similar games. I installed ds4drv, but when I run "ds4drv" he returned errors:

`
Traceback (most recent call last):
  File "/home/ddd/.venv/bin/ds4drv", line 5, in <module>
    from ds4drv.__main__ import main
  File "/home/ddd/.venv/lib/python3.12/site-packages/ds4drv/__main__.py", line 5, in <module>
    from .actions import ActionRegistry
  File "/home/ddd/.venv/lib/python3.12/site-packages/ds4drv/actions/__init__.py", line 1, in <module>
    from ..action import ActionRegistry
  File "/home/ddd/.venv/lib/python3.12/site-packages/ds4drv/action.py", line 1, in <module>
    from .config import add_controller_option
  File "/home/ddd/.venv/lib/python3.12/site-packages/ds4drv/config.py", line 72, in <module>
    class Config(configparser.SafeConfigParser):
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?

`

If I run "sudo ds4drv" he returned this: 

`
sudo ds4drv 
Traceback (most recent call last):
  File "/usr/local/bin/ds4drv", line 33, in <module>
    sys.exit(load_entry_point('ds4drv==0.5.1', 'console_scripts', 'ds4drv')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/ds4drv", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/local/lib/python3.12/dist-packages/ds4drv-0.5.1-py3.12.egg/ds4drv/__main__.py", line 6, in <module>
  File "/usr/local/lib/python3.12/dist-packages/ds4drv-0.5.1-py3.12.egg/ds4drv/actions/__init__.py", line 1, in <module>
  File "/usr/local/lib/python3.12/dist-packages/ds4drv-0.5.1-py3.12.egg/ds4drv/action.py", line 1, in <module>
  File "/usr/local/lib/python3.12/dist-packages/ds4drv-0.5.1-py3.12.egg/ds4drv/config.py", line 72, in <module>
AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
`

I tried cross pipx (sudo pipx install ds4drv) and created .venv (python3 -m venv .venvm / source .venv/bin/activate / sudo pip install ds4drv)  where used common pip, but he returned similar errors

Thanks for help!

raahimfareed, 2025-02-06:
https://github.com/raahimfareed/ds4drv

I just changed a single file to make it work
Since the package is not maintained anymore, it's using an outdated name

You can either use my changed repo or change it yourself by going to `config.py` file, finding `SafeConfigParser` and replacing it with `ConfigParser`

SpiningMonkey, 2025-04-20:
> https://github.com/raahimfareed/ds4drv
>
> I just changed a single file to make it work Since the package is not maintained anymore, it's using an outdated name
>
> You can either use my changed repo or [...]

Hello, ı have the same ıssue. I want to try this but it's in root file and ı can't access it. İs there a way to access this file? Thanks for help.

raahimfareed, 2025-06-13:
> Hello, ı have the same ıssue. I want to try this but it's in root file and ı can't access it. İs there a way to access this file? Thanks for help.

I don't have access to this right now but

If it's root ownership, you can use sudo to change it,
Something like `sudo nano /path/to/file`

Although, be careful since changing any wrong file can break your system, make sure it's just the ds4drv file that you are changing