chrippa/ds4drv #214: AttributeError

afverschuur, 2025-02-22:
Hi!

I connected my PS4 Controller using bluetoothctl (BleuZ 5.66) to my Raspberry Pi4 (OS Lite), installed ds4drv using pip in virtual environment (python3.11-full). Added rules for permissions as described, reloaded the rules, got 'Failed to create  input device "dev/uinput" cannot be opened for writing'. Loaded uinput (sudo modprobe uinput). And when I run $ ds4drv I get:

Traceback (most recent call last):
  File "/home/afverschuur/venv/bin/ds4drv", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/__main__.py", line 151, in main
    thread = create_controller_thread(index + 1, controller_options)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/__main__.py", line 120, in create_controller_thread
    controller = DS4Controller(index, controller_options, dynamic=dynamic)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/__main__.py", line 35, in __init__
    self.load_options(self.options)
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/__main__.py", line 93, in load_options
    self.fire_event("load-options", options)
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/__main__.py", line 38, in fire_event
    self.loop.fire_event(event, *args)
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/eventloop.py", line 86, in fire_event
    self.process_events()
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/eventloop.py", line 92, in process_events
    callback(*args)
  File "/home/afverschuur/venv/lib/python3.11/site-packages/ds4drv/actions/input.py", line 84, in load_options
    joystick.device.device.fn)
    ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'InputDevice' object has no attribute 'fn'. Did you mean: 'fd'?

Is it something I did wrong? Or is it a bug?

Ravgni, 2025-03-01:
Same issue on Raspberri Pi3 (OS Lite). VEnv with python3.11. Ds4drv installed via pip (0.5.1)

nisel11, 2025-03-04:
Try changing fn to fd in /usr/lib/python3/site-packages/ds4drv/actions/input.py via terminal and restarting the service

xeluior, 2025-03-08:
> Try changing fn to fd in input.py via terminal and restarting the service

Came here with the same issue. This works.

RedBearAK, 2025-04-24:
@nisel11 @xeluior @Ravgni @afverschuur 

> Try changing fn to fd in input.py via terminal and restarting the service

From what I understand "fd" is for the file descriptor (which are just integers), and the proper replacement for the deprecated "fn" attribute (probably short for "filename") is the "path" attribute.

sgpinkus, 2025-04-26:
`pip install evdev==1.8` worked for me, or at least gets past this error. Now it just does this blocking forever (not sure if that's right ...):

```
$ ds4drv --hidraw
[info][controller 1] Created devices /dev/input/js2 (joystick) /dev/input/event20 (evdev) 
[info][hidraw] Scanning for devices
````